greatscottgadgets/hackrf #992: Need Scheme

BondJamesB, 2021-11-12:
### Steps to reproduce
Dear,
I have used a wrong charger for my product.
Wich ended up blowing a fuse.
I need a scheme for replacement.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2021-11-12:
I'm sorry. I don't quite understand your request, but I will do my best to answer. I think you are asking for the schematics. If so, the schematics are in this repository.

I'm also not certain what you mean by charger as the HackRF doesn't require a charger. 

Can you please post a picture of your HackRF board (front and back) and your set up?

lyusupov, 2021-11-21:
https://github.com/rad1o/hardware/raw/master/datasheets/else/hackrf-one-schematic.pdf

closed 2021-12-06 by straithe