greatscottgadgets/hackrf #997: Opera Cake LED indicators using time switching mode

maxroseGIT, 2021-11-19:
I tried testing the time switching mode suggested as in #931. As there is no documentation yet about the LEDs on the Opera Cake board (see #530), I am not sure whether my observations of the LEDs point to a hardware issue or a software issue. 

Any help or suggestions would be very welcome.

I soldered the board myself and by hand. Had a hard time to solder the marked pieces:

[image]

But overall it should work.

### Steps to reproduce
```
$ hackrf_operacake -o 0 -m time -t A1:1000 -t A2:1000 -t A3:1000 -t A4:1000                                                                                                                                                  
$ hackrf_transfer -r /dev/null
```

### Expected behaviour
LEDs of the specified ports should light up (check attached picture)

[image]

### Actual behaviour
LEDs of bank B light up although I didn't call them in the above command. And moreover they do not light up synchronous to bank A, which would suggest some systematic error. 

### Version information
**skywave linux 4.1**:
Base System: Ubuntu 20.04 LTS

**hackrf_info output:**
```
hackrf_info version: git-80948a2
libhackrf version: git-80948a2 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000223c69dc2299344f
Board ID Number: 2 (HackRF One)
Firmware Version: git-80948a2 (API:1.05)
Part ID Number: 0xa000cb3c 0x005d474d
Opera Cake found, address: 0
```

miek, 2021-11-19:
In time switching mode the B ports should match the A ports, so this does point to a possible hardware issue.

Are you able to switch the correct ports using manual mode? e.g:
```
$ hackrf_operacake -m manual
$ hackrf_operacake -a A1 -b B1
$ hackrf_operacake -a A2 -b B2
$ hackrf_operacake -a A3 -b B3
$ hackrf_operacake -a A4 -b B4
```

maxroseGIT, 2021-11-19:
Thanks for the suggested manual mode check. I tried and found that LEDs A1-4 light up as expected. B1 and B3 do also what is expected but B2 and B4 do not light up at all. So I guess this is an issue that was produced by soldering the Opera Cake board by myself.

```
$ hackrf_operacake -m manual
$ hackrf_operacake -a A1 -b B1    // A1 and B1 LED light up
$ hackrf_operacake -a A2 -b B2    // A2 and B1 LED light up
$ hackrf_operacake -a A3 -b B3    // A3 and B3 LED light up
$ hackrf_operacake -a A4 -b B4    // A4 and B3 LED light up
```

maxroseGIT, 2021-11-19:
What are actually the two LEDs for? The LED marked with a red circle obviously indicates the selected Port. But what is the LED marked with the green circle is for?

[image]

miek, 2021-11-19:
> What are actually the two LEDs for? The LED marked with a red circle obviously indicates the selected Port. But what is the LED marked with the green circle is for?
> 

They're intended to be populated as two different colours, along with the single LEDs by the PA0/PB0 ports. PA0 should be green and PB0 yellow. Then the different colours light up to indicate which side they're connected to.
So, for example: if A2 lights up green then it's connected to A0, or if B3 lights up yellow it's connected to B0.

miek, 2021-11-19:
> Thanks for the suggested manual mode check. I tried and found that LEDs A1-4 light up as expected. B1 and B3 do also what is expected but B2 and B4 do not light up at all. So I guess this is an [...]

OK, that narrows it down some. I suspect that the U3CTRL0 line (labelled in the schematic) may be shorted to ground, so I'd check the soldering on U4, U5, U9, U11, U13, and U15.

maxroseGIT, 2021-11-19:
Thanks @miek for the suggested lines to look at.

Indeed I have found a shorted solder point at U11 on the U3CTRL0 line.

[image]

I used flux and soldered it properly on its seperated pads. I didn't clean up the flux yet.

The ports do now correctly light up upon selection.

Thanks for the immediate help!

closed 2021-11-19 by maxroseGIT