greatscottgadgets/hackrf #999: Rx Tx problem with amp/ signal is gone

Mopar44084, 2021-11-21:
### Steps to reproducce
1. Use hackrf with sdrsharp or other app
2. Tune to frequency or transmit
3. Turn amp to 1

### Expected behaviour
Tell us what you expect should happen
Amplifies signal

### Actual behaviour
Tell us what happens instead
Signal is gone or will not transmit
Vga/LNA is weak without amp also

### Version information
**Operating system**:
HamPi os RPI 3
Skywave Linux - Ubuntu 16.04 - laptop
**hackrf_info output:**
Mayhem 1.4.1 - original portapack h1

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2021-11-29:
Can you show me what your waterfall looks like before and after you enable the amp?

Mopar44084, 2021-11-29:
The first 2 pictures are from mayhem 1.4.1 
Second 2 pictures and video is from cubicsdr on skywave Linux. This is a local radio station, it used to have a lot more red in the waterfall, now I have to crank up LNA and VGA to even get a signal
[image]
[image]
[image]
[image]

Mopar44084, 2021-11-29:
The first 2 pictures are from mayhem 1.4.1 
Second 2 pictures is from CubicSDR on Skywave Linux. This is a local radio station, it used to have a lot more red in the waterfall, now I have to crank up LNA and VGA to even get a signal
[image]
[image]
[image]
[image]

Mopar44084, 2021-11-29:
This is using CubicSDR on Skywave Linux
https://user-images.githubusercontent.com/86167080/143959091-b0dd1c4a-2c76-4619-b6da-89c6c8cb1fd4.mp4

straithe, 2021-11-30:
Thanks for the extra information. I'm a bit limited in how I can help with your current setup as I don't know much if anything about CubicSDR, Skywave Linux, or Mayhem. To clarify, it seems that you primarily want help assessing if the HackRF hardware is ok. Is that correct?

Mopar44084, 2021-11-30:
I believe there may be a hardware problem, when amp is selected it cuts the signal and overall I have noticed a reduction in signal. When I tx in amp mode it also cuts the signal.

straithe, 2021-12-03:
That's an interesting idea! Can you first post a picture of your HackRF (front and back) here?

Mopar44084, 2021-12-04:
[image]
#
[image]
[image]

Mopar44084, 2021-12-04:
[image]
[image]
[image]

straithe, 2021-12-06:
If you tried to send a picture of the back of the HackRF it didn't come through. Could you post again?

Mopar44084, 2021-12-07:
[image]

Mopar44084, 2021-12-07:
Here is the back, I don't know why it wouldn't send?
[image]

gozu42, 2021-12-07:
wth github? 
i can see @Mopar44084 trying to repost the backside image on discord #github-updates ... 
... but i cant see the posts here? 
not even if i go to the specific URL in the notifications, and it doesnt seem like a client-side caching issue either.

Mopar44084, 2021-12-07:
I am not sure why it didn't send, sorry!
[image]

straithe, 2021-12-12:
I don't see any obvious faults with the board, but it does sound like you've encountered a hardware problem, so I suggest contacting the reseller you purchased the board from. As long as you've operated the board within the specifications in our documentation (https://hackrf.readthedocs.io/en/latest/hackrf_one.html#transmit-power) they should be able to help you out!

closed 2021-12-12 by straithe

Mopar44084, 2021-12-12:
Thanks! Is there anything I need to tell them? I bought it from nooelect.

straithe, 2021-12-12:
Nothing specific. Going through their customer support should be sufficient.

Mopar44084, 2021-12-17:
I sent Nooelec 2 mails and left a message and they don't seem to respond, anything else I can do?

straithe, 2021-12-18:
Can you reach out to the info@greatscottgadgets.com and include a link to this ticket?

Mopar44084, 2021-12-18:
I just sent one out, along with my receipt from nooelec.

Thank you!

Mopar44084, 2022-04-11:
Is there a way you can check into my RMA info at nooelec?
I got an email in February 23 that it would take 4-12 weeks.

Thanks a lot!

straithe, 2022-04-12:
You're still within that 12 week period so please check back once that 12 weeks has elapsed.