greatscottgadgets/hackrf #1000: I have no clue how to set up the Opera Cake

0xGSMonkey, 2021-11-25:
I have no clue what to do to set up the opera cake after I built it.

I keep getting "Operacakes found: None" error

I am not sure what I am supposed to do in order to get it to work.

gozu42, 2021-11-25:
what version of hackrf firmware + userspace are you using?
please post the output of "hackrf_info".

0xGSMonkey, 2021-11-25:
Lol, it could be that I solder some things together wrong. I just thought about that. I don't have the LEDs soldered on. 

But, requested info:
hackrf_info version: git-5f1536c*
libhackrf version: git-5f1536c*(0.6)
Firmware Version: 2018.01.1 (API:1.02)

I took out the Index, serial number, board id and part id.

gozu42, 2021-11-25:
that userland is from 2021-04, the firmware is ... a bit older.
can you please update both to current git head?
there has been at least one change related to operacake since april, and even though i dont think it matters here, it would help if everyone involved is staring at the same code while trying to figure this out.

0xGSMonkey, 2021-11-26:
hackrf_info version: git-26c0f70
libhackrf version: git-26c0f70 (0.6)
Firmware Version: git-26c0f70 (API:1.05)

root@user-Space-Shuttle-888:/home/user/Hacking/hackrf/firmware/hackrf_usb/build# hackrf_operacake -l
Opera Cakes found: None

I really believe I might have messed up putting the board together. It shorts the board out when I stack Opera Cake on top of the HackRF One. Unless there It isn't shorting out the HackRF

gozu42, 2021-11-26:
great.
if you want someone to take a look at your soldering, please post reasonably high resolution pictures of your board.
"shorts the board out" doesnt sound too good, really.

but we can continue a bit without that too.
if you look at the code for "hackrf_operacake -l" [here](https://github.com/greatscottgadgets/hackrf/blob/22267f3b8e71bd064337921444b0e40509f47b43/host/hackrf-tools/src/hackrf_operacake.c#L343), it basicly just calls hackrf_get_operacake_boards, for which the other end is [here](https://github.com/greatscottgadgets/hackrf/blob/22267f3b8e71bd064337921444b0e40509f47b43/firmware/hackrf_usb/usb_api_operacake.c#L30), and that just returns the currently-known-to-the-firmware list of operacakes.
that list gets populated during operacake_init [here](https://github.com/greatscottgadgets/hackrf/blob/22267f3b8e71bd064337921444b0e40509f47b43/firmware/common/operacake.c#L115), and is basicly just some i2c probing.
which means if the list is empty, that probing fails, meaning the hackrf firmware fails to talk (in the right way) to the PCA9557PW on the operacake board.
so, besides investigating the general "shorting out the board" situation, i would recommend to focus on that for now.

0xGSMonkey, 2021-11-26:
I got it. I think. But, if I have any more problems I will definitely let everyone know. If this thread can stay open for the next couple of weeks I would appreciate that. But, yeah my soldering definitely needs some work. One of the capacitors on the board is not on the pad at all. I am going to try to fix that this week if I can. 

hackrf_info:
hackrf_info version: git-5f1536c*
libhackrf version: git-5f1536c*(0.6)
Firmware Version: git-22267f3 (API:1.05) [+]

[+] this branch(?) seemed to fix it. or I got super lucky. And it just didn't short.

After compiling from github source and not using the tar.bz2 file I was able to get opera cake to work.

gozu42, 2021-11-26:
note that in this last post your hackrf_info+libhackrf versions have gone back to the 2021-04 version (5f1536c), i would recommend you stick to the "git head" version (26c0f70) you had 2h earlier for anything operacake related.
(there really has been a lot of work on operacake in the last months)

for reworking the soldering and testing the result, recommend you start with the i2c connection, and then work towards the sma sockets, if that makes any sense. also, populating the LEDs means you get visual feedback on what the switches think they are doing, which might help a lot. 

keeping the issue open for a while is probably no problem, straithe is _supposed_ to be on vacation and not closing it, but i also thought it meant not _labeling_ issues. but, even if you or someone else closes it, you can always just reopen it if another question pops up.

for slightly more interactive support, you could also join #hackrf on [the GSG discord](https://discord.gg/rsfMw3rsU8).

straithe, 2021-11-29:
I'm happy to keep this issue for a while. The advice given by @gozu42 is sound. 

I will be adding a lot more documentation on Opera Cake over the next while as Great Scott Gadgets ramps up production. ☺️

straithe, 2021-12-28:
@0xGSMonkey do you need any more assistance, or may I close this issue?

straithe, 2022-01-28:
I'm going to close this as there hasn't been a response in a while, but please open a new issue if you need further assistance.

closed 2022-01-28 by straithe