greatscottgadgets/hackrf #1001: External clock not recognized

realtimepeople, 2021-11-27:
### Steps to reproduce

1. 10 MHz CMOS 3.3V Input to CLKIN (from GPS DXO)
2. Check "hackrf_debug -s -n o -r" delivers : "[ 0] -> 0x01"
3. Cross-check without inut to CLKIN : delivers 0xd1 (or 0x51 after reset)

### Expected behaviour

Clock output and operation shall be synchronous to the external clock signal.

### Actual behaviour

Slightly different frequency on clock output compared to clock input.
Receiver not locked to the external clock.

### Version information

**Operating system**:
Linux lbox 5.4.0-90-lowlatency #101-Ubuntu SMP PREEMPT Fri Oct 15 20:57:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

**hackrf_info output:**

hackrf_info version: git-26c0f70*
libhackrf version: git-26c0f70* (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000406464c8237a2c4b
Board ID Number: 2 (HackRF One)
Firmware Version: git-22267f3 (API:1.05)
Part ID Number: 0xa000cb3c 0x0076434e

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

Gnuradio + osmosdr used to check coherency

martinling, 2021-11-30:
Hi @realtimepeople, thanks for the report.

I've tested on a HackRF running the same `26c0f70` firmware version, with CLKIN coming from a HackRF with TCXO, and watching both CLKIN and CLKOUT of the receiving unit on the oscilloscope. The two clocks did sync up, but only after making a `hackrf_transfer -r` run first, and then re-enabling the clock output with `hackrf_clock -o 1` (the CLKOUT disappeared at first for some reason).

Could you check if doing this works for you?

Something that you may not have spotted is this note in the [docs](https://hackrf.readthedocs.io/en/latest/hackrf_one.html#external-clock-interface-clkin-and-clkout):
> The switch to or from CLKIN only happens when a transmit or receive operation begins.

So at startup it's actually normal to see `hackrf_debug -s -n 0 -r` return `0x01` without the clock having synced yet, since the switch to external clock input only happens when TX or RX is started.

However, the part where the CLKOUT disappeared when running `hackrf_transfer` was unexpected, so we're treating that as a firmware bug.

realtimepeople, 2021-12-03:
I have checked the coherency with a simple GRC scheme:
The 10MHz sine from my GPSDO as Hackrf input, an extra
3.3V 10MHz signal derived form the GPSDO to the EXT clock.
So RX operation was active.

No lock to the external CLK.
No phase-coherency between Hackrf's CLKOUT and the GPSDO.

Unfortunately my 10MHz Schmitt-Trigger quit, so I can continue 
testing earliest after fixing my hardware...

realtimepeople, 2021-12-03:
Correction:

My hardware is fixed.
Now I have an even more confusing observation:

Hackrf's CLKIN and CLKOUT are coherent in phase.
But the reception is not coherent to the external clock!
See ultra-short [video](https://schaefer.eit.h-da.de/2021-12-03%2019-28-08.mkv)

realtimepeople, 2021-12-04:
Sorry, my conclusion was wrong.

After switching to the external clock Hackrf runs
smoothly and with very little phase noise guided
by the external clock.
But: Because of the noise at intermediate frequency (IF)  = 0 
it is necessary  to choose a software IF somewhere around 1kHz.
In my previous test I used 1 MHz, which was way too high.

Sorry to waste your time.

martinling, 2021-12-04:
You didn't waste our time! Amongst other things, we discovered a firmware bug (the CLKOUT being disabled as a side-effect of the firmware detecting and switching to CLKIN). I'm going to leave this issue open for now as a reminder for us to tackle that (or at least open a separate issue for it).

We were discussing this issue today and we wondered whether the lack of coherency you saw could be due to the actual tuned frequency not being exactly 10MHz. The way the firmware chooses the LO and IF frequencies is complicated and doesn't always result in an exact match for the center frequency requested. So we were going to suggest using explicit tuning.

realtimepeople, 2021-12-04:
Exactly that: To be really coherent I needed to know the exact LO frequency I get
when I want some arbitrary number as input frequency. 
Knowing this I can tune the "software second LO"
to mix the wanted frequency down to exactly zero.
I could not find the documentation about the exact frequency tuning by now.
Is there a documentation page explaining it ?

martinling, 2021-12-07:
> I could not find the documentation about the exact frequency tuning by now.
> Is there a documentation page explaining it ?

Explicit tuning is not particularly well documented at the moment. It can be used in `hackrf_transfer`, and from programs using libhackrf directly, but other programs may not make the necessary options accessible.

For hackrf_transfer, explicit tuning can be specified by the `-i`, `-o` and `-m` options, which are briefly described in the help output but may not make much sense without knowing the HackRF architecture.

The key thing to know is that HackRF uses an IF between 2150 and 2750MHz, and has a mixer in front of the IF stage for which the LO is from 84MHz to 5400MHz. The mixer can be either bypassed, used with a high-pass image rejection filter, or used with a low-pass one. Those three settings - IF, LO and filter/bypass selection - define the tuning.

In general, to be sure of ensuring that a perfect match for a desired frequency is possible, you might also need to look at the datasheets for the parts involved and the available register settings (which can be read out or set using `hackrf_debug`).

martinling, 2021-12-07:
I've branched off two new issues from this one - one for the CLKOUT bug, and one for better documentation of explicit tuning.

mossmann, 2021-12-07:
Adding to @martinling's comment: The LO for the RF mixer is in steps of approximately (but not exactly) 1 MHz, and the LO for the IF mixer is in steps of about 28.61 Hz.

When you've tuned the HackRF to 9 MHz in the video, the HackRF has tuned as close as possible to 9 MHz, but there is up to 28.61 Hz of tuning error. You should be able to observe that the frequency error remains consistent over time because it is phase locked. If you remove the clock input, the frequency error will drift over time.

mossmann, 2021-12-07:
By playing with different explicit tuning options as @martinling suggested, you may end up with a different amount of frequency error, but there isn't a way to eliminate that error. There will always be a tuning error between 0 and 28.61 Hz.

realtimepeople, 2021-12-07:
Sure, I have immediately recognized the tuning error.
But this can be eliminated with the software local oscillator.
My observed remaining frequency offset is around (delta f) / f = 10^-11,
observed over 1 hour !
Hackrf ist GREAT, especially when it is locked to a frequency standard :-)

martinling, 2021-12-07:
> there isn't a way to eliminate that error. There will always be a tuning error between 0 and 28.61 Hz.

Well, it should be possible to eliminate that error if you're flexible about what centre frequency you use, because for some of them that tuning error will be zero. It would be nice to know which frequencies can have that property. Particularly any that are also an integer number of Hz.

mossmann, 2021-12-07:
It is theoretically possible to tune to certain frequencies such that the frequency error is zero, but we don't have a practical way to do so (at least not without some firmware development) outside of the 2150 to 2750 MHz band.

realtimepeople, 2021-12-07:
I used GQRX to read the chosen LO frequency after tuning to a specific frequency.
Then I just needed some slight additional software tuning in my GQRX blockset to achieve zero beat frequency.
Somewhat empirical but pretty useful.

mossmann, 2021-12-07:
I'm so glad that is working for you, @realtimepeople! May we close this issue?

realtimepeople, 2021-12-07:
YUP - this job is done.
I will now go searching for a way to tag samples precisely using the 1PPS pulse from the GNSS.
THANKS !

mossmann, 2021-12-07:
Nice! We don't currently have a facility in firmware to help with timestamping, but if you come up with something we would love to see it!

closed 2021-12-07 by mossmann