greatscottgadgets/hackrf #1002: Airlines.db update

ssthomassen, 2021-11-28:
Hi,

I want to know how I can update the Airlines.db file? Cant find any documentation for that file, but it looks kind of simple to update?

First alot of 3 letter 'words', and the the description, but do i have to write all down alphabetic?

straithe, 2021-11-29:
I'm sorry. I don't know what Airlines.db is referring to. I've searched the HackRF code and can't find any reference to what you are talking about. Are you perhaps confusing the HackRF code with the [mayhem code](https://github.com/eried/portapack-mayhem)?

ssthomassen, 2021-11-29:
Thanks for your answer @straithe , I thought the HackRF was the basic, and mayhem was a layer on top. But it can be that i should ask someone with mayhem knowledge. The Airlines.db is the list for the ADB-S view, to see airlines company.

straithe, 2021-11-29:
I don't know much about Mayhem, so I don't have the skill to help you further. I suggest you ask your question on the Mayhem repository I linked in my last message.

closed 2021-11-29 by straithe