greatscottgadgets/hackrf #1003: Samplerate changes cause IQ glitch in stream

gibbi, 2021-11-30:
### Steps to reproduce
1. constantly change the samplerate of a running stream between two different rates
2. after usually between 20 - 100 ratechanges the "IQIQIQ .." data gets shifted to "QIQIQIQI .." samples (a one byte shift in the stream)
3. after the same random duration it glitches back into a working state
4. if no further rate change happens the misalignment persists 

### Expected behaviour
this should not happen

### Actual behaviour
the same physical channel gets mapped to I in one run and Q in the next.

### Version information
(16,18,20.04) Ubuntu
(2018, 2021, latest firmware/library)

### Output
[image]

in the setup i had lose cables connected to the I+ and I- pads, rendering this channel noisy while the Q channel was shorted. This shows how the same physical channel turned form Q to I.

this is exactly the same issue as #439

mossmann, 2021-11-30:
Thank you! This is a very clear test case. I'm not certain of the cause, but I suspect it is the way the sample clock on HackRF One is produced by dividing a 2x sample clock by two (so that we have a 2x clock for the Double Data Rate interface).

I doubt that we considered mid-stream sample rate modifications when designing that part of the system. Can you tell us about your use case? Would it be a problem for you if we were to disallow mid-stream sample rate changes?

gibbi, 2021-11-30:
we really need the ratechange and have to work around this. restarting the streamer kind of works but this also crashes eventually in a way only a reset will fix. i will play around in the firmware see what i can do. if that fails we have to go for a watch dog that just resets the hackrf whenever its stuck. any hint about how the components interact would be great.

straithe, 2021-11-30:
Thanks for highlighting your need for the ratechange. Can you give more details about your use case so we can try to recreate the situation and/or confirm that any solutions we try will address your concerns? I'm also curious about whether disallowing mid-stream sample rate changes would cause issues for you. Thoughts?

gibbi, 2021-12-01:
@mossmann @straithe I solved the problem as far as I can tell, please review if this might have any side effects

straithe, 2021-12-01:
Good to hear! Can you tell us how you went about solving the problem? I'd appreciate the extra insight.

martinling, 2021-12-01:
@straithe the fix is in #1004 - I'm just reviewing it now.

PR #1004 (2021-12-02): fix stream glitch on rate change
this changes fixes #1003 by disabling cpld processing while changing the samplingrate
closed 2021-12-02 by martinling