greatscottgadgets/hackrf #1007: [FreeBSD] host/hackrf-tools: cmake set_release problem

cederom, 2021-12-03:
### Steps to reproduce
1. Pull latest head master.
2. Build `host/hackrf-tools` on FreeBSD.
3. At `mkdir build; cd build; cmake ..` see cmake error.

### Expected behaviour
CMake runs with no errors.

### Actual behaviour
CMake errors.

### Version information
**Operating system**: `FreeBSD 0xCFMX4 13.0-STABLE FreeBSD 13.0-STABLE #0 stable/13-n247979-3e322ded35f: Thu Nov  4 12:28:58 CET 2021     root@0xCFMX4:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64`.

**hackrf_info output:** does not build.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
% uname -a
FreeBSD 0xCFMX4 13.0-STABLE FreeBSD 13.0-STABLE #0 stable/13-n247979-3e322ded35f: Thu Nov  4 12:28:58 CET 2021     root@0xCFMX4:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

% cc --version
FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Target: x86_64-unknown-freebsd13.0
Thread model: posix
InstalledDir: /usr/bin

% cmake --version
cmake version 3.21.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).

% pwd
XXX/hackrf.git/host/hackrf-tools

% git branch
* master

% git log -3 --oneline
d60fb83 (HEAD -> master, origin/master, origin/HEAD) Cleanup of host software CMake build system (#664)
e9dd7eb Merge pull request #1004 from gibbi/AG-FixRateChange
fc8b3c1 fix stream glitch on rate change

% mkdir build
% cd build
% cmake ..
-- The C compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:28 (include):
  include could not find requested file:

    set_release

CMake Warning (dev) at CMakeLists.txt:29 (add_definitions):
  Policy CMP0005 is not set: Preprocessor definition values are now escaped
  automatically.  Run "cmake --help-policy CMP0005" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.21)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "XXX/hackrf.git/host/hackrf-tools/build/CMakeFiles/CMakeOutput.log".
```

straithe, 2021-12-03:
@multiplemonomials would you be willing to take a look at this?

miek, 2021-12-03:
A workaround for this is to start from the `host/` directory and `mkdir build; cd build; cmake ..` from there, instead of from `host/hackrf-tools/`. It looks like building from further down the tree was broken by #664.

mossmann, 2021-12-03:
Assuming we fix this and want to continue supporting builds from the `hackrf-tools` and `libhackrf` directories, we should add those builds to CI.

cederom, 2021-12-03:
> A workaround for this is to start from the `host/` directory and `mkdir build; cd build; cmake ..` from there, instead of from `host/hackrf-tools/`. It looks like building from further down the tree was broken by #664.

Yup :-) Confirmed :-)

```
% git branch
* master

% pwd
/XXX/hackrf.git/host/build

% mkdir build
% cd build
% cmake ..
-- The C compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking dependencies...
-- Looking for fftw_execute in libfftw3.so
-- Looking for fftw_execute in libfftw3.so - found
-- Looking for fftwf_execute in libfftw3f.so libfftw3.so
-- Looking for fftwf_execute in libfftw3f.so libfftw3.so - found
-- Found FFTW: /usr/local/lib/libfftw3.so
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "1.8.0")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.13
-- Looking for libusb_init in libusb.so
-- Looking for libusb_init in libusb.so - found
-- Found USB1: /usr/lib/libusb.so
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test CMath_LINKS_WITHOUT_M
-- Performing Test CMath_LINKS_WITHOUT_M - Failed
-- Performing Test CMath_LINKS_WITH_M
-- Performing Test CMath_LINKS_WITH_M - Success
-- Found CMath: /usr/lib/libm.so
-- Configuring done
-- Generating done
-- Build files have been written to: /XXX/hackrf.git/host/build

% make
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf-static.dir/hackrf.c.o
[ 10%] Linking C static library libhackrf.a
[ 10%] Built target hackrf-static
[ 15%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.o
[ 20%] Linking C shared library libhackrf.so
[ 20%] Built target hackrf
[ 25%] Building C object hackrf-tools/src/CMakeFiles/hackrf_sweep.dir/hackrf_sweep.c.o
[ 30%] Linking C executable hackrf_sweep
[ 30%] Built target hackrf_sweep
[ 35%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 40%] Linking C executable hackrf_debug
[ 40%] Built target hackrf_debug
[ 45%] Building C object hackrf-tools/src/CMakeFiles/hackrf_clock.dir/hackrf_clock.c.o
[ 50%] Linking C executable hackrf_clock
[ 50%] Built target hackrf_clock
[ 55%] Building C object hackrf-tools/src/CMakeFiles/hackrf_info.dir/hackrf_info.c.o
[ 60%] Linking C executable hackrf_info
[ 60%] Built target hackrf_info
[ 65%] Building C object hackrf-tools/src/CMakeFiles/hackrf_cpldjtag.dir/hackrf_cpldjtag.c.o
[ 70%] Linking C executable hackrf_cpldjtag
[ 70%] Built target hackrf_cpldjtag
[ 75%] Building C object hackrf-tools/src/CMakeFiles/hackrf_spiflash.dir/hackrf_spiflash.c.o
[ 80%] Linking C executable hackrf_spiflash
[ 80%] Built target hackrf_spiflash
[ 85%] Building C object hackrf-tools/src/CMakeFiles/hackrf_operacake.dir/hackrf_operacake.c.o
[ 90%] Linking C executable hackrf_operacake
[ 90%] Built target hackrf_operacake
[ 95%] Building C object hackrf-tools/src/CMakeFiles/hackrf_transfer.dir/hackrf_transfer.c.o
[100%] Linking C executable hackrf_transfer
[100%] Built target hackrf_transfer

% ./hackrf-tools/src/hackrf_info
hackrf_info version: git-d60fb83
libhackrf version: git-d60fb83 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f660ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cbXX 0x00XX4XX6

% ./hackrf-tools/src/hackrf_spiflash -w ../../../hackrf-2021.03.1/firmware-bin/hackrf_one_usb.bin
File size 35444 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 35444 bytes at 0x000000.

% ./hackrf-tools/src/hackrf_info
hackrf_info version: git-d60fb83
libhackrf version: git-d60fb83 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f660ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cbXX 0x00XX4XX6

<<<RESET>>>

% ./hackrf-tools/src/hackrf_info
hackrf_info version: git-d60fb83
libhackrf version: git-d60fb83 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f660ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cbXX 0x00XX4XX6

```

IT'S ALIVE AND FRESH! :-)

cederom, 2021-12-03:
Hmm, the whole `host` bundle needs to be built at once, as noted in https://github.com/greatscottgadgets/hackrf/issues/1008, when building only the `host/hackrf-tools` part it will use parts installed in the system and also may lead to build problems.

Thank you guys! :-)

multiplemonomials, 2021-12-04:
Yeah, when I made #664 I did not operate under the assumption that one would try to build just a subdirectory, that isn't a very standard thing to do in CMake.  If we want to support that it'll need some extra changes.  Cool to see that it works on FreeBSD though which I didn't test it on!

closed 2021-12-13 by miek