greatscottgadgets/hackrf #1008: [FreeBSD] hackrf-2021.03.1/host/hackrf-tools: use of undeclared identifier

cederom, 2021-12-03:
### Steps to reproduce
1. Get 2021.03.1 release, extract.
2. Build `host/hackrf-tools`.
3. See build problems related to undefined variables.

### Expected behaviour
Builds fine.

### Actual behaviour
Does not build. Probably some headers or header location missing?

### Version information
**Operating system**:

**hackrf_info output:** Does not build.

### Output
```
% uname -a
FreeBSD 0xCFMX4 13.0-STABLE FreeBSD 13.0-STABLE #0 stable/13-n247979-3e322ded35f: Thu Nov  4 12:28:58 CET 2021     root@0xCFMX4:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

% cc --version
FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Target: x86_64-unknown-freebsd13.0
Thread model: posix
InstalledDir: /usr/bin

% cmake --version
cmake version 3.21.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).

% pwd
/XXX/hackrf-2021.03.1/host/hackrf-tools

% mkdir build
% cd build
% cmake ..
CMake Deprecation Warning at CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "1.8.0")
-- Found LIBHACKRF: /usr/local/lib/libhackrf.so
-- Found FFTW: /usr/local/lib/libfftw3.so
-- Configuring done
-- Generating done
-- Build files have been written to: /XXX/hackrf-2021.03.1/host/hackrf-tools/build

% make
[  6%] Building C object src/CMakeFiles/hackrf_sweep.dir/hackrf_sweep.c.o
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:110:24: error: use of undeclared identifier 'SAMPLES_PER_BLOCK'
uint32_t num_samples = SAMPLES_PER_BLOCK;
                       ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:113:22: error: use of undeclared identifier 'MAX_SWEEP_RANGES'
uint16_t frequencies[MAX_SWEEP_RANGES*2];
                     ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:234:11: error: use of undeclared identifier 'BYTES_PER_BLOCK'
                        buf += BYTES_PER_BLOCK;
                               ^
/XXX/hackrf/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:259:50: error: use of undeclared identifier 'SAMPLES_PER_BLOCK'
                                        (uint64_t)(num_samples + THROWAWAY_BLOCKS * SAMPLES_PER_BLOCK)
                                                                                    ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:270:11: error: use of undeclared identifier 'BYTES_PER_BLOCK'
                        buf += BYTES_PER_BLOCK;
                               ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:274:11: error: use of undeclared identifier 'BYTES_PER_BLOCK'
                        buf += BYTES_PER_BLOCK;
                               ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:278:10: error: use of undeclared identifier 'BYTES_PER_BLOCK'
                buf += BYTES_PER_BLOCK - (fftSize * 2);
                       ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:433:7: error: use of undeclared identifier 'MAX_SWEEP_RANGES'
                        if(MAX_SWEEP_RANGES <= num_ranges) {
                           ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:436:7: error: use of undeclared identifier 'MAX_SWEEP_RANGES'
                                                MAX_SWEEP_RANGES);
                                                ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:512:20: error: use of undeclared identifier 'SAMPLES_PER_BLOCK'
        if (num_samples % SAMPLES_PER_BLOCK) {
                          ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:517:20: error: use of undeclared identifier 'SAMPLES_PER_BLOCK'
        if (num_samples < SAMPLES_PER_BLOCK) {
                          ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:668:11: warning: implicit declaration of function 'hackrf_init_sweep' [-Wimplicit-function-declaration]
        result = hackrf_init_sweep(device, frequencies, num_ranges, num_samples * 2,
                 ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:669:38: error: use of undeclared identifier 'INTERLEAVED'
                        TUNE_STEP * FREQ_ONE_MHZ, OFFSET, INTERLEAVED);
                                                          ^
/XXX/hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:676:12: warning: implicit declaration of function 'hackrf_start_rx_sweep' [-Wimplicit-function-declaration]
        result |= hackrf_start_rx_sweep(device, rx_callback, NULL);
                  ^
2 warnings and 12 errors generated.
*** Error code 1

Stop.
make[2]: stopped in /XXX/hackrf-2021.03.1/host/hackrf-tools/build
*** Error code 1
```

mossmann, 2021-12-03:
Might there be a mismatch between the installed libhackrf version and the version of hackrf-tools you are trying to compile? You should be able to avoid this problem by building both at once (from the host directory instead of the hackrf-tools directory).

cederom, 2021-12-03:
True! Thank you @mossmann I will build the whole `tools` bundle :-)

```
# pkg delete hackrf
Checking integrity... done (0 conflicting)
Deinstallation has been requested for the following 2 packages (of 0 packages in the universe):

Installed packages to be REMOVED:
        SoapyHackRF: 0.3.3_2
        hackrf: 2017.02.1

Number of packages to be removed: 2

Proceed with deinstalling packages? [y/N]: y
[1/2] Deinstalling SoapyHackRF-0.3.3_2...
[1/2] Deleting files for SoapyHackRF-0.3.3_2: 100%
[2/2] Deinstalling hackrf-2017.02.1...
[2/2] Deleting files for hackrf-2017.02.1: 100%

% pkg info hackrf
pkg: No package(s) matching hackrf

% pwd
/XXX/hackrf-2021.03.1/host

% rm -rf build
% mkdir build
% cd build
% cmake ..
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Deprecation Warning at libhackrf/CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PkgConfig: /usr/local/bin/pkg-config (found version "1.8.0")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.13
CMake Warning (dev) at /usr/local/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIBUSB)
  does not match the name of the calling package (USB1).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindUSB1.cmake:39 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  libhackrf/CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found LIBUSB: /usr/lib/libusb.so
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
CMake Deprecation Warning at hackrf-tools/CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found FFTW: /usr/local/lib/libfftw3.so
-- Configuring done
-- Generating done
-- Build files have been written to: /XXX/hackrf-2021.03.1/host/build

% make
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf-static.dir/hackrf.c.o
[ 10%] Linking C static library libhackrf.a
[ 10%] Built target hackrf-static
[ 15%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.o
[ 20%] Linking C shared library libhackrf.so
[ 20%] Built target hackrf
[ 25%] Building C object hackrf-tools/src/CMakeFiles/hackrf_sweep.dir/hackrf_sweep.c.o
[ 30%] Linking C executable hackrf_sweep
[ 30%] Built target hackrf_sweep
[ 35%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 40%] Linking C executable hackrf_debug
[ 40%] Built target hackrf_debug
[ 45%] Building C object hackrf-tools/src/CMakeFiles/hackrf_clock.dir/hackrf_clock.c.o
[ 50%] Linking C executable hackrf_clock
[ 50%] Built target hackrf_clock
[ 55%] Building C object hackrf-tools/src/CMakeFiles/hackrf_info.dir/hackrf_info.c.o
[ 60%] Linking C executable hackrf_info
[ 60%] Built target hackrf_info
[ 65%] Building C object hackrf-tools/src/CMakeFiles/hackrf_cpldjtag.dir/hackrf_cpldjtag.c.o
[ 70%] Linking C executable hackrf_cpldjtag
[ 70%] Built target hackrf_cpldjtag
[ 75%] Building C object hackrf-tools/src/CMakeFiles/hackrf_spiflash.dir/hackrf_spiflash.c.o
[ 80%] Linking C executable hackrf_spiflash
[ 80%] Built target hackrf_spiflash
[ 85%] Building C object hackrf-tools/src/CMakeFiles/hackrf_operacake.dir/hackrf_operacake.c.o
[ 90%] Linking C executable hackrf_operacake
[ 90%] Built target hackrf_operacake
[ 95%] Building C object hackrf-tools/src/CMakeFiles/hackrf_transfer.dir/hackrf_transfer.c.o
[100%] Linking C executable hackrf_transfer
[100%] Built target hackrf_transfer

% ./hackrf-tools/src/hackrf_info
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f660ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00574766
```

closed 2021-12-03 by cederom