greatscottgadgets/hackrf #1010: hackrf_sweep_spectrum_analyzer not working with 2021.03.1 firmware?

pmcculey, 2021-12-06:
### Steps to reproduce
1. Install the 2021.03.1 firmware
2. Run hackrf_sweep_spectrum_analyzer.jar 

### Expected behaviour
Works with 2017.02.1 firmware but not with the newest version. 

### Actual behaviour
hackrf_sweep_spectrum_analyzer loads, the RX LED on the board lights, but no spectra are displayed on PC. Re-flashing the 2017.02.1 firmware restores functionality as far as hackrf_sweep_spectrum_analyzer operation is concerned

### Version information
**Operating system**: Win10x64

**hackrf_info output:** 
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc27746ec3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x008f474e

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

gozu42, 2021-12-06:
> libhackrf version: git-cc7f599 (0.5)
> Firmware Version: 2021.03.1 (API:1.04)

my guess is pretty much this being the problem. this combines 2017 userland/libhackrf with 2021 firmware.
can you try to update the libhackrf version to 2021 too?

pmcculey, 2021-12-06:
First line in the release notes, don't know how I missed that!! Thanks for pointing it out. Next adventure is to find out how to update libhackrf. I'd be grateful for any any pointers. I'm at the bad end of the learning curve on this sadly.

dmaltsiniotis, 2021-12-06:
> I'm at the bad end of the learning curve on this sadly.

No worries, this should be easy one to fix: The Pothos folks have handily bundled the latest hackrf tools and libraries as part of their project. If you download the latest Pothos SDR software, it will come with the latest versions of the hackrf libraries and tools.

The documentation on the process is here: https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#windows-binaries

And the download area is here: https://downloads.myriadrf.org/builds/PothosSDR/ just select the top/latest version.

There is the option of compiling/building from source code as well, but I'm assuming here you're looking for just a download link to the latest version.

pmcculey, 2021-12-06:
I have the latest PothosSDR stuff from that link already installed in C:\PothosSDR (it was not working when installed in the Program Files folder....). However I also had GNU radio 3.8 installed & it seems the libhackrf was being picked up from there maybe? I uninstalled GNU radio & now hackrf_info gives me:

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc27746ec3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x008f474e

The hackrf_sweep_spectrum_analyzer is still not working though. I should have mentioned earlier that in the hackrf_sweep_spectrum_analyzer GUI, I see a message saying that HackRF is disconnected even thouhg starting that caused the RX LED to light.

Thanks for the help! I'm at least making some progress.

gozu42, 2021-12-06:
not sure that is going to work. 
afair hackrf_sweep_spectrum_analyzer is using its own version of libhackrf (that is wrapped in "interesting" ways), so you could try to open an issue there to get them to update that to 2021.
besides that, you are probably stuck with using a firmware that is old enough.

pmcculey, 2021-12-07:
Yes, hackrf_sweep_spectrum_analyzer seems to work with the 2018 firmware. However its not necessairly recommended to update to that version as the only difference between it and the 2017 version is improvements when using the DFU bootloader. The 2021 firmware seems to have more updates and bugfixes. Aside from that I see that using hackrf_sweep with the older firmware does not produce output! When I issue "hackrf_sweep.exe -f 10:110 -w 8192", I get an error as follows:
"hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)"
Invoking it with the same arguments using the 2021 firmware works as I'd expect, Another thing to investigate.....

I could be stuck with the 2017 firmware if I want to use the GUI.

dmaltsiniotis, 2021-12-07:
Throwing this out there: if you're goal is just to visualize the hackrf_sweep output, I have tested that https://github.com/xmikos/qspectrumanalyzer works on Windows 10 (as well as MacOS) with the latest firmware and libraries. It may be worth exploring rather than fighting with hackrf_sweep_spectrum_analyzer.

pmcculey, 2021-12-07:
I'll give that a try, thanks. If hackrf_sweep_spectrum_analyzer operates in some sort of non-standard way as suggested, it's unlikely to be any use to me. It's useful for a quick look and has a nice UI, but there is no way to dump the data to disk which was going to be the next problem.

dmaltsiniotis, 2021-12-07:
> It's useful for a quick look and has a nice UI, but there is no way to dump the data to disk which was going to be the next problem.

qspecturmanalyzer just calls the hackrf_sweep process under the covers. You can call this process directly yourself as well if you wanted to get at the raw data. It has a couple modes of operation, include on that writes to a CSV file for processing the data later, and one mode that operations in a binary format which is intended more for inter-process communication (such as between qspecturmanalyzer and hackrf_sweep.exe for real-time visualziation.)

So one options would be to use the GUI to verify things are working as expected, then craft your own command line to write the data you need to disk.

```
# hackrf_sweep -h 
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .

```

Cheers,

DM

straithe, 2022-01-28:
@pmcculey , I'm following up on this issue. Has it been resolved?

pmcculey, 2022-01-28:
I've had to park it due to pressures of other work. I was able to get an
executable version that worked, but would like to get the python code
operational so I could customise it for my needs. I want to build an
autonomous RFI monitor for our radio observatory.

Is it OK for me to contact you when I can get back at it?

Thanks,

Joe

straithe, 2022-02-09:
Yes, please feel free to re-open this issue or open another when you are able to return to this project.

closed 2022-02-09 by straithe