greatscottgadgets/hackrf #1013: CLKOUT is disabled when CLKIN is detected

martinling, 2021-12-07:
### Steps to reproduce
1. Use `hackrf_clock -o 1` to enable CLKOUT.
2. Connect a 10MHz clock source to CLKIN.
3. Start reception with `hackrf_transfer -r /dev/null`.

### Expected behaviour
CLKOUT signal remains active, and syncs to CLKIN.

### Actual behaviour
CLKOUT signal disappears.

After re-enabling CLKOUT with `hackrf_clock -o 1`, the signal reappears, and further `hackrf_transfer` calls do not make it disappear again, so I suspect that CLKOUT is being disabled as a side-effect of the CLKIN detection and switchover.

Spotted whilst reproducing #1001.

mossmann, 2022-02-08:
I'm having difficulty reproducing this. Did you reproduce, @gozu42?

mossmann, 2022-02-08:
> I'm having difficulty reproducing this. Did you reproduce, @gozu42?

Oops. My bad. I had the wrong serial number in my `hackrf_transfer` command. I can reproduce exactly as @martinling described.

gozu42, 2022-02-08:
(seeing you found it by now, but posting anyways since i was 2/3 done)

just whatever firmware is on DUT right now, i think this one is bug-916 head
```
Firmware Version: git-7057235a (API:1.05)
```

starting situation, no external clock
```
$ hackrf_debug --si5351c -n 0 -r
[  0] -> 0x51
$ hackrf_clock -r 3
        clock control = Down, Int Mode, PLL src A, XTAL, 2 mA
```

enabling clkout, verifying it is enabled
```
$ hackrf_clock -o 1
$ hackrf_clock -r 3
        clock control = Up, Int Mode, PLL src A, MULTISYNTH SELF, 8 mA
```

plugging in a tcxo on P22
```
$ hackrf_debug --si5351c -n 0 -r
[  0] -> 0x01
$ hackrf_clock -r 3
        clock control = Up, Int Mode, PLL src A, MULTISYNTH SELF, 8 mA
```

some rx, and recheck clkout
```
$ hackrf_transfer -r /dev/null -n 1000000
$ hackrf_clock -r 3
        clock control = Down, Int Mode, PLL src A, XTAL, 2 mA
```