greatscottgadgets/hackrf #1021: Hackrf firmware update fail

oldnewham, 2021-12-16:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

oldnewham, 2021-12-16:
I tried to update the firmware on my HackRF and the update failed, the board now appears dead.
only the power led comes on and the rx/tx leds but they are very dim, the usb rf  iv8 not not light up
I have tried everything possible to get it working. Board no 25940 the board is only 3 months old.

lyusupov, 2021-12-16:
https://github.com/greatscottgadgets/hackrf/issues/976

gozu42, 2021-12-16:
not a whole lot of detail here currently, and the little there is sounds _very_ different from #976 

can you please run "dmesg -Tw", then try a DFU boot (press and hold DFU button, press and release reset button, release DFU button" and paste what happens on the dmesg during that.

also, can you please provide pictures of the front and back of your hackrf board?

closed 2022-01-28 by straithe