greatscottgadgets/hackrf #1024: hardware not recognized by any device

cyb3rfr0gg, 2021-12-28:
### Steps to reproduce
1.tried on raspberry pi4
2.tried on dell corei 7 
3.linux console

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead
hardawared to being detected
### Version information
**Operating system**:
no detected

**hackrf_info output:**

not detected 
If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

gozu42, 2021-12-28:
try running "sudo dmesg -Tw" in a console (this prints the current system log buffer, then continously updates it as new messages come in, you can stop it by hitting ctrl-c).
then plugging in the hackrf. 
any reaction in the dmesg output?

keep the dmesg running.
try a DFU boot by keeping the hackrf plugged in, press+hold the DFU button, press+release the reset button, release the DFU button.
any dmesg reaction to that?

any LED activity on the hackrf on either?
did you try a different USB cable already?

cyb3rfr0gg, 2021-12-28:
sudo dmesg -Tw gives me this 
[Tue Dec 28 17:28:59 2021] Linux version 5.14.0-kali4-amd64 (devel@kali.org) (gcc-10 (Debian 10.3.0-12) 10.3.0, GNU ld (GNU Binutils for Debian) 2.37) #1 SMP Debian 5.14.16-1kali1 (2021-11-05)
[Tue Dec 28 17:28:59 2021] Command line: BOOT_IMAGE=/boot/vmlinuz-5.14.0-kali4-amd64 root=UUID=fd2928b1-8733-4e83-963a-9ad20ccc2c6f ro quiet splash
[Tue Dec 28 17:28:59 2021] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[Tue Dec 28 17:28:59 2021] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[Tue Dec 28 17:28:59 2021] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[Tue Dec 28 17:28:59 2021] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[Tue Dec 28 17:28:59 2021] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[Tue Dec 28 17:28:59 2021] signal: max sigframe size: 1776
[Tue Dec 28 17:28:59 2021] BIOS-provided physical RAM map:
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x0000000000100000-0x000000007ffeffff] usable
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x000000007fff0000-0x000000007fffffff] ACPI data
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[Tue Dec 28 17:28:59 2021] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[Tue Dec 28 17:28:59 2021] NX (Execute Disable) protection: active
[Tue Dec 28 17:28:59 2021] SMBIOS 2.5 present.
[Tue Dec 28 17:28:59 2021] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[Tue Dec 28 17:28:59 2021] Hypervisor detected: KVM
[Tue Dec 28 17:28:59 2021] kvm-clock: Using msrs 4b564d01 and 4b564d00
[Tue Dec 28 17:28:59 2021] kvm-clock: cpu 0, msr 5a201001, primary cpu clock
[Tue Dec 28 17:28:59 2021] kvm-clock: using sched offset of 15841071296 cycles
[Tue Dec 28 17:28:59 2021] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[Tue Dec 28 17:28:59 2021] tsc: Detected 2594.000 MHz processor
[Tue Dec 28 17:28:59 2021] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[Tue Dec 28 17:28:59 2021] e820: remove [mem 0x000a0000-0x000fffff] usable
[Tue Dec 28 17:28:59 2021] last_pfn = 0x7fff0 max_arch_pfn = 0x400000000
[Tue Dec 28 17:28:59 2021] Disabled
[Tue Dec 28 17:28:59 2021] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[Tue Dec 28 17:28:59 2021] CPU MTRRs all blank - virtualized system.
[Tue Dec 28 17:28:59 2021] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[Tue Dec 28 17:28:59 2021] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[Tue Dec 28 17:28:59 2021] RAMDISK: [mem 0x2e7eb000-0x333ecfff]
[Tue Dec 28 17:28:59 2021] ACPI: Early table checksum verification disabled
[Tue Dec 28 17:28:59 2021] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[Tue Dec 28 17:28:59 2021] ACPI: XSDT 0x000000007FFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[Tue Dec 28 17:28:59 2021] ACPI: FACP 0x000000007FFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[Tue Dec 28 17:28:59 2021] ACPI: DSDT 0x000000007FFF0470 002325 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[Tue Dec 28 17:28:59 2021] ACPI: FACS 0x000000007FFF0200 000040
[Tue Dec 28 17:28:59 2021] ACPI: FACS 0x000000007FFF0200 000040
[Tue Dec 28 17:28:59 2021] ACPI: APIC 0x000000007FFF0240 00005C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[Tue Dec 28 17:28:59 2021] ACPI: SSDT 0x000000007FFF02A0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[Tue Dec 28 17:28:59 2021] ACPI: Reserving FACP table memory at [mem 0x7fff00f0-0x7fff01e3]
[Tue Dec 28 17:28:59 2021] ACPI: Reserving DSDT table memory at [mem 0x7fff0470-0x7fff2794]
[Tue Dec 28 17:28:59 2021] ACPI: Reserving FACS table memory at [mem 0x7fff0200-0x7fff023f]
[Tue Dec 28 17:28:59 2021] ACPI: Reserving FACS table memory at [mem 0x7fff0200-0x7fff023f]
[Tue Dec 28 17:28:59 2021] ACPI: Reserving APIC table memory at [mem 0x7fff0240-0x7fff029b]
[Tue Dec 28 17:28:59 2021] ACPI: Reserving SSDT table memory at [mem 0x7fff02a0-0x7fff046b]
[Tue Dec 28 17:28:59 2021] No NUMA configuration found
[Tue Dec 28 17:28:59 2021] Faking a node at [mem 0x0000000000000000-0x000000007ffeffff]
[Tue Dec 28 17:28:59 2021] NODE_DATA(0) allocated [mem 0x7ffc6000-0x7ffeffff]
[Tue Dec 28 17:28:59 2021] Zone ranges:
[Tue Dec 28 17:28:59 2021]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[Tue Dec 28 17:28:59 2021]   DMA32    [mem 0x0000000001000000-0x000000007ffeffff]
[Tue Dec 28 17:28:59 2021]   Normal   empty
[Tue Dec 28 17:28:59 2021]   Device   empty
[Tue Dec 28 17:28:59 2021] Movable zone start for each node
[Tue Dec 28 17:28:59 2021] Early memory node ranges
[Tue Dec 28 17:28:59 2021]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[Tue Dec 28 17:28:59 2021]   node   0: [mem 0x0000000000100000-0x000000007ffeffff]
[Tue Dec 28 17:28:59 2021] Initmem setup node 0 [mem 0x0000000000001000-0x000000007ffeffff]
[Tue Dec 28 17:28:59 2021] On node 0, zone DMA: 1 pages in unavailable ranges
[Tue Dec 28 17:28:59 2021] On node 0, zone DMA: 97 pages in unavailable ranges
[... cut ...]
[Tue Dec 28 17:29:27 2021] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[Tue Dec 28 17:29:31 2021] RRScreenChangeNotify event received
[Tue Dec 28 17:29:31 2021] Monitor 0 (w,h)=(1920,914) (x,y)=(0,0)
[Tue Dec 28 17:29:31 2021] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[Tue Dec 28 17:29:31 2021] RRScreenChangeNotify event received
[Tue Dec 28 17:29:31 2021] Monitor 0 (w,h)=(1920,914) (x,y)=(0,0)
[Tue Dec 28 17:29:31 2021] Sending monitor positions (8 of them)  to the host: VINF

cyb3rfr0gg, 2021-12-28:
all led ligths seems to be ok 
any command  i can run to test the hardware 
 did not gave m,e version

gozu42, 2021-12-28:
that is quite verbose, but mostly not the interesting parts. 
much more interesting is what dmesg says when you connect the hackrf, or try the DFU boot.
it is also a virtual linux, so it might be better to do the dmesg test on the raspi if you dont have a non-virtual linux otherwise.

cyb3rfr0gg, 2021-12-28:
let me try the pi 
 just a moment

cyb3rfr0gg, 2021-12-28:
on the Pi i got same  results 
any other command you want me to try 
?

gozu42, 2021-12-28:
same result means what exactly?
is there _any_ reaction in the "dmesg -Tw" output if you plug in the hackrf or try a DFU boot?

could you please post pictures of the front+back of the hackrf board?

cyb3rfr0gg, 2021-12-28:
sure  i got it from nooelec off amazon  paid around $380  is original 
when i  try the Software Radio Spectrum Analyzer on GRC the console shows empty device addrss , no devices found 
 
[image]
[image]
[image]

gozu42, 2021-12-28:
lets try to see if the system sees the hackrf first, before any more complex software.
so, if you run dmesg -Tw on the raspi, and plug in the hackrf, do you get any new output?
if not, how about the DFU boot?

if dmesg reports seeing the hackrf in non-DFU boot, what happens if you run hackrf_info?

cyb3rfr0gg, 2021-12-28:
it said version unknowkn and no boards found

cyb3rfr0gg, 2021-12-28:
[image]

cyb3rfr0gg, 2021-12-28:
so , what you think should i request a returning mailing label before is too late ?

martinling, 2021-12-28:
Hi @cyb3rfr0gg,

Please try a DFU boot:

With the HackRF connected to USB, hold the DFU button down, and while holding it, press and release the reset button.

You should see the pattern of LEDs change when you do this.

Then run `dmesg` again and see if there are any new messages at the end. You can also run `lsusb` and look for a line like:

`1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)`

cyb3rfr0gg, 2021-12-29:
im not sure but if this is a good sign 
[image]

straithe, 2021-12-29:
Looks great!

cyb3rfr0gg, 2021-12-29:
ok at elast some progress after few days of research thsi is the 1st good news i got thnx ,now i study cybersecurity and  wanted this hackrf for pentest , listen to radio and stuff like that , im still getting errors on the grc console that there is no device , im new to sdr but i like the challenge  , just need some guidance on how to make this thing work 
[image]
[image]
[image]

straithe, 2021-12-29:
https://hackrf.readthedocs.io/en/latest/ and https://www.greatscottgadgets.com/sdr/ are great places to start. You may also want to join the GSG Discord: https://discord.gg/rsfMw3rsU8

miek, 2021-12-29:
You can't use the "UHD: USRP Source" block for HackRF, that block is for a different product.
Either the "osmocom Source" or "Soapy Source" blocks work with HackRF.

cyb3rfr0gg, 2021-12-30:
ok thanks ,i really tried to follow up with the GSG lessons but since GRC
calls the blocks a different name is hard , im researching a lot online but
have not really seen the device in action , if you know any links with
projects for beginners , as i mentioned before i study cybersecurity and
would like to  use hack rf as a pentesting tool ,  to listen to radio
communications nearby ... at least i want to know that the hackrf one is
working then i can relax and  learn how to play with it , but since i have
not use it yet  im worried that i may have to return it , please point me
in the right direction here and  thanks , so far the CS is being great

cyb3rfr0gg, 2021-12-30:
Hi there do you have any suggestions on any projects i can work on  , i'm
having a hard time to follow up the tutorials  , i just don't want to
return the device  but i'm clueless , i like the challenge but i feel maybe
this not a product that i can benefit from, any help  is appreciated,
thanks again bye

straithe, 2021-12-30:
A bunch of videos to watch for inspiration are here: https://wctf.us/resources . Also, please join the GSG Discord. There are lots of people to talk to and get ideas from.

cyb3rfr0gg, 2021-12-30:
thanks i'm going to do that right now

straithe, 2022-01-28:
I'm going to close this as there hasn't been an update in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2022-01-28 by straithe

cyb3rfr0gg, 2022-01-28:
Thanks for the follow up , i finally got the hackrf tested , i was able to
do the replay attack and tuned the device to listen local radio frqz,
sorry if i was impatient, just did not want3d to loose the returning window
, now that 8 know is working the potential of this tool im very happy ,
thanks again ,