greatscottgadgets/hackrf #1031: HackRF One no work and very very hot

sastre, 2022-01-07:
Hi people. 
I have a problem with my HackRF One. I will try to explain the problem
- I have been using it during some years with SDR#: all OK. 
- A few years ago, I bought the Portapack H2 with HAVOC firmware: all OK
- 4 days ago, I tried to update the HAVOC to MAYHEM firmware. 
- My PC correctly detected but I could not get the update because my version of Windows is old (apparently ...)
- I confirm you my USB cable work OK and I have not made any hardware modifications. Only try to update the firmware
- After what I have described, when I had to use the USB cable to charge HackRF One+Portapack H2, the device do not well
- Sometimes the LCD mean with light or the blue Led of rotary encoder mean On. 
- Then, I disconnected the HackRF one from Portapack H2 to check both devices to check it separately.
- Finally, I discovered that, after 1 sec or less after connecting the USB cable, the main IC (LPC4320 ...) reaches a very very very high temperature.
- If I switch on with the Portapack H2 with your internal LiPo battery, (without USB cable) the exact same thing happens.

 I can not belive...
Both devices always work OK. Together or alone. But now, no work
Could someyone help me?
```

gozu42, 2022-01-08:
lets debug this in steps, starting with just the hackrf.
please remove the portapack from the hackrf.
press+hold the DFU button, plug in the hackrf (or press+release reset if already plugged in), release the DFU button.
do you see an NXP dfu device on the host after this?

do you have any linux around, even if its just a random raspi or so? (in my exp, that makes usb debugging somewhat easier)

also, could you please post pictures of the front and back of the hackrf board?

sastre, 2022-01-08:
Hi! The portapack already disconnected from the HackRF One. When I switch
on with the USB cable the HackRF One not work and the IC LPC4320... gets
very very hot in 1 second only.
I have used DFU and RESET buttons but the problem remind. I have not Linux.
Only I have W10 32bits.
I think that there is some electronic component of HackRF One damaged ...

straithe, 2022-01-10:
Which version of the Portapack do you have? Can you add a picture here?

sastre, 2022-01-10:
I use Portapack H2 (see photos). Until a few days ago, I always use HackRF
One with Portapack H2 Havoc firmware HAVOC with no problem

straithe, 2022-01-10:
I do not see any photos. If you are responding to this GitHub issue from email, attachments will not be seen. 

Can you make sure you get pictures of all USB ports while your HackRF is in the Portapack?

sastre, 2022-01-10:
And now?

straithe, 2022-01-10:
Can you please open your issue in GitHub and respond with pictures via the website and not through your email?

sastre, 2022-01-11:
OK. Here the photos of may portapack H2
[image]
[image]
[image]

closed 2022-01-11 by sastre