greatscottgadgets/hackrf #1032: Problem with HackRF and RationalWaves

jaeger-audiodesign, 2022-01-08:
My HackRF works under Win10 with SDR-Console on my MacBook with Parallels and on my ThinkPad.
I installed the app Rational Waves on both devices. The App detects the HackRF on both devices the RX LED light up when I start scanning but I get no data. Has anyone an idea?

On my HackRF is the actual firmware installed.

straithe, 2022-01-10:
I don't quite understand what question you are asking, so I have some clarifying questions. Is there a particular reason you are using Rational Waves? Are you specifically asking for help with Rational Waves, or are you asking a question about your HackRF? Any extra detail would be appreciated.

jaeger-audiodesign, 2022-01-10:
The main problem was, that the HackRF didn't work under Windows10. No I found out, that this was an issue with the newest Firmware. I made a downgrade to the 2017 FW Version and now I get data in the hackRF_sweep app. I suspect that the RationalWaves software works on the basic of the hackRF_sweep tool. I wrote the support and wait for a feedback...

straithe, 2022-02-09:
Which support did you write to?

closed 2022-04-29 by straithe