greatscottgadgets/hackrf #1036: hackrf won't transmit bellow 25 MHz

otpisani, 2022-01-14:
I can't transmit with hackrf bellow 25MHz with build in utility on linux arch. That would be issue in hardware like faulty amp, or differences in firmware an build hackrf_tools. I'm not sure if tools and firmware match.

hackrf_transfer -s 2 -f 20608000 -x 0 -a 0 -t testing_20608

gozu42, 2022-01-14:
my guess is the problem may be on the receive side.
while an rtlsdr v3 dongle does support these frequencies, they require the use of the "direct sampling" mode.
see https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/

otpisani, 2022-01-14:
I know that, and it's set for HF mode in-phase (I) in SDR console, other frequency bellow 30MHz can receive with v3 rtl dongle

gozu42, 2022-01-14:
> HF mode in-phase (I) in SDR console

note the rtlsdr v3 user guide explicitly calls that ... "useless".
try selecting direct sampling of Q instead.

miek, 2022-01-14:
The command you posted is setting the sample rate to 2Hz (`-s 2`) which is well below minimum supported rate, I suggest trying with a rate of at least 2MHz.

otpisani, 2022-01-14:
> 

If I choose Q direct sampling I get "wider" bandwidth with in particular frequency, with I sampling the same frequency looks thinner, but nothing is change regarding not transmit anything under 25.5 MHz, no matter if I flash latest or firmware from 2018, also I try another USB port like switching from USB3 to legacy USB2, and hackrf is "only" USB2, didn't change anithying with that, also I try boot with ISO image, like kali, didn't help also, basically I don't have any idea left, except I saw some yt clip when guy experiment transmit with AngelSDR, but don't know how to configure software, and don't know any other GUI tool which support transmit, maybe I try with GNU radio

otpisani, 2022-01-14:
> The command you posted is setting the sample rate to 2Hz (`-s 2`) which is well below minimum supported rate, I suggest trying with a rate of at least 2MHz.

-s 2 means 2 MHz, 10 is default, but nothing changes, also if I don't put -s parameter, default is used with 0 results

straithe, 2022-01-28:
> The command you posted is setting the sample rate to 2Hz (`-s 2`) which is well below minimum supported rate, I suggest trying with a rate of at least 2MHz.

Mike's comment is correct. Can you please attempt updating your command and letting me know the results, @otpisani ?

otpisani, 2022-01-28:
> > The command you posted is setting the sample rate to 2Hz (`-s 2`) which is well below minimum supported rate, I suggest trying with a rate of at least 2MHz.
> 
> @miek 's comment is correct. Can you please attempt updating your command and letting me know the results, @otpisani ?

Hello, Yes @miek is right, 2 is Hz, and 20000000 is 20MHz, but can't see nothing in waterfall with 20000000 Hz also , but with 35000000 Hz can see big spike on 13.8MHz with around -62dBm, so if I'm not doing anything wrong with -s sample size this time, question is what can I do wrong regarding hackrf one health if I enter something like this 
hackrf_transfer -f 13800000 -x 47 -s 35000000 -t some.raw

martinling, 2022-01-28:
> hackrf_transfer -f 13800000 -x 47 -s 35000000 -t some.raw

35000000 Hz (35MHz) is not a supported sample rate. The maximum sample rate of HackRF One is 20MHz.

otpisani, 2022-01-28:
> > hackrf_transfer -f 13800000 -x 47 -s 35000000 -t some.raw
> 
> 35000000 Hz (35MHz) is not a supported sample rate. The maximum sample rate of HackRF One is 20MHz.

Short video with sample 35MHz
https://www.dropbox.com/s/0d1j50kyo0lp6e6/35000000-2022-01-28_17.44.09.mp4?dl=0

martinling, 2022-01-28:
You are not achieving a sample rate of 35MHz in that video. The HackRF is trying to, because you've asked it to, but it cannot not work, because that is nearly twice the supported sample rate of the device,.

Even putting aside that the SGPIO pipeline cannot keep up with 35MHz and the DACs are not rated to 35MHz, you are not even getting all that data to the HackRF. The data rate at a 35MHz sample rate is 70MB/s (each sample is two bytes - one for I and one for Q). As you can see from the `hackrf_transfer` output, you are only achieving 47.7MB/s, because the USB bus becomes saturated at that point. So nearly a third of your samples are just being lost.

As you can see from the waterfall in your video, this results in garbage all over the spectrum when you transmit.

otpisani, 2022-01-28:
> You are not achieving a sample rate of 35MHz in that video. The HackRF is trying to, because you've asked it to, but it cannot work, because that is nearly twice the supported sample rate of the device.
> 
> Even putting aside that the SGPIO pipeline cannot keep up with 35MHz and the DACs are not rated to 35MHz, you are not even getting all that data to the HackRF. The data rate at a 35MHz sample rate is 70MB/s (each sample is two bytes - one for I and one for Q). As you can see from the `hackrf_transfer` output, you are only achieving 47.7MB/s, because the USB bus becomes saturated at that point. So nearly a third of your samples are just being lost.
> 
> As you can see from the waterfall in your video, this results in garbage all over the spectrum when you transmit.

Yap, everything is correct that you said, I'm getting between 39.8 and 40.6 MiB/sec with max sample 20MB, so USB bus eta is aprox 100%, but can't see anything on waterfall until -s sample is 33MHz, command is

hackrf_transfer -f 13800000 -x 47 -R -s 20000000 -t ~/Downloads/rtl-sdr/test.raw > can't see on waterfall spike

hackrf_transfer -f 13800000 -x 47 -R -s 33000000 -t ~/Downloads/rtl-sdr/test.raw > can see on waterfall spike with power around -82dBm

Input file end reached. Rewind to beginning.
132.6 MiB / 1.000 sec = 132.6 MiB/second
47.4 MiB / 1.000 sec = 47.4 MiB/second
Input file end reached. Rewind to beginning.
47.7 MiB / 1.000 sec = 47.7 MiB/second
47.4 MiB / 1.000 sec = 47.4 MiB/second
Input file end reached. Rewind to beginning.
47.7 MiB / 1.000 sec = 47.7 MiB/second
47.4 MiB / 1.000 sec = 47.4 MiB/second
47.7 MiB / 1.000 sec = 47.7 MiB/second

What Am I  doing wrong...

martinling, 2022-01-28:
OK, please stop trying to overclock the HackRF because it's not going to help with anything. You're just making it transmit nonsense all over the spectrum when you do that.

It doesn't make any sense to be trying out different sample rates for the same input file. The sample rate you transmit with needs to correspond to the sample rate of the signal data that you're trying to transmit. How did you make the `test.raw` file? What is supposed to be in it, at what sample rate?

otpisani, 2022-01-28:
> OK, please stop trying to overclock the HackRF because it's not going to help with anything. You're just making it transmit nonsense all over the spectrum when you do that.
>
> It doesn't make any [...]

test.raw is made with sox something like this 

sox input.wav output.raw

basically it's just stream header info from wav file, and wav file it's made with sdr console, 

file input.waw output 

26-Jan-2022 141336.153 14.424MHz 005.wav: RIFF (little-endian) data, WAVE audio, Microsoft PCM, 16 bit, stereo

martinling, 2022-01-28:
OK, you're missing a few steps. You cannot create an audio transmission by just feeding an audio file into `hackrf_transfer`. That's not what the tool is for.

The input to `hackrf_transfer` is raw I/Q samples, at a sample rate of 2-20MHz. Those samples define the actual radio waves you want to transmit, not an audio signal that you want to send over them.

If you want to transmit an audio signal, you'd need to decide how you're going to modulate that audio (e.g. AM or FM, and the parameters for each), and generate a modulated I/Q signal from the audio. That would then be something that you could transmit with `hackrf_transfer`.

Typically though, you would do this sort of thing in GNU Radio (in which case you don't need to use `hackrf_transfer` at all).

You may find it helpful to work through some or all of our [video series](https://greatscottgadgets.com/sdr/).

otpisani, 2022-01-28:
> OK, you're missing a few steps. You cannot create an audio transmission by just feeding an audio file into `hackrf_transfer`. That's not what the tool is for.
>
> The input to `hackrf_transfer` is raw [...]

Thanks for that, but shortly, how can I "catch" something with hackrf_transfer like this 

hackrf_transfer -f xy Hz -r some_test.raw

AND than transmit that raw sample back

hackrf_transfer -f xy Hz -t some_test.raw

beacuse can't see anything with default samples, which is 10MHz

martinling, 2022-01-28:
`hackrf_transfer` isn't intended to be a replay tool.

To replay a signal correctly requires getting quite a few things right. You would need to capture the target signal with appropriate levels of RX, IF and baseband gains, separate it from surrounding signals with a suitable filter bandwidth, and then retransmit it with an suitable gain level. There are other issues to think about as well.

I strongly recommend watching [lesson 11 of our video series](https://greatscottgadgets.com/sdr/11/), which is specifically about replay.

If you just run the two commands you've shown above, then you'll be indiscriminately capturing and replaying everything within +/-5MHz of the center frequency `xy`, without any control of gain levels.

otpisani, 2022-01-28:
> `hackrf_transfer` isn't intended to be a replay tool.
> 
> To replay a signal correctly requires getting quite a few things right. You would need to capture the target signal with appropriate levels of RX, IF and baseband gains, separate it from surrounding signals with a suitable filter bandwidth, and

Thats would be the reason why I try to "catch" target signal with sdr v3 dongle on win platform, with direct sampling mode (without up converter) is better tool for HF frequency than hackrf one, hackrf one is great too no question about itl, but not for HF band, or without filter.

 > then retransmit it with an suitable gain level. There are other issues to think about as well.
> I strongly recommend watching [lesson 11 of our video series](https://greatscottgadgets.com/sdr/11/), which is specifically about replay.
>
> If you just run the two commands you've shown above, then [...]

Presume because of 10MHz sample rate, +-5MHz from "peak" frequency!!!???, that would be great jamming device with extra gain power. 

Great answer btw, maybe you can elaborate why I can transmit with success something above 25MHz,  with this two commands from above...

Case is closed for me, I will watch this video, thanks for advice.

closed 2022-01-28 by straithe