greatscottgadgets/hackrf #1038: HackRF One only 3v3 Light

0xGSMonkey, 2022-01-24:
### Steps to reproduce
1.  Connected Pin 11 and Pin 18 on P22
2. Disconnect Pin 11 and Pin 18 on P22
3.

### Expected behaviour
Only the 3v3 Light is on and sometimes the RF light.

### Actual behaviour

I don't think it is the USB cable I have used 4 different USB cables and all of them show the 3V3 light as being the only light that shows. I am not sure what happened. I connected PIn 11 and PIn 18 on Expansion Interface 22 without reading the docs. Did this fry my hackrf one? D:

gozu42, 2022-01-24:
shorting VCC to GND seems like a good way to fry things, yes.
can you measure the voltage between pins 11+18 on P22?

and if i may ask, why did you connect these pins to begin with? (as in, what were you trying to do?)
and just to add the standard request: can you please post pics of the front+back of your hackrf board?

0xGSMonkey, 2022-01-24:
I added pins to the operacake in order to use it as an add-on board I think those two pins connected on the operacake board and possibly fried the hackrf one. 

Between Pin 11 and Pin 18 on P22 the volage is 3.25. Is there a datasheet or something on what it should be? I did a visual inspection of the board and there doesn't seem to be any damage, but here is a picture of the board. 

It still only seems to show that the 3v3 light is the only light that comes on. 

[image]
[image]<br />

straithe, 2022-01-31:
Are you able to bring up the HackRF with DFU mode?

closed 2022-03-07 by straithe