greatscottgadgets/hackrf #1043: HackRF with Portapack H1 suddenly does not connect to computer via USB

goa, 2022-02-01:
Hello! 

I have a HackRF with Portapack H1. It was working perfectly until yesterday, when it stopped connecting to the computer via USB. 

• I've tested multiple cables and tried it on MacOS and Windows 10 (two windows computers actually), but nothing seems to help. It was working perfectly just a few hours ago.

• I've tried setting it into DFU mode, but again, it is not discoverable as a USB device on either OS, and the `dfu-util-static.exe` cannot find it on windows.

• Besides the USB connection, the device is powered up by USB and all Mayhem firmware functions seem to work perfectly via the PortaPack H1. 

• The 3V3, 1V8 and RF leds, all light up (but not the USB led).

### Steps to reproduce
1. Connect to computer via USB.
2. Enter HackRF mode via the Mayhem firmware menu.

### Expected behaviour
The USB led should light up,  and the device should be discoverable by windows or MacOS.

### Actual behaviour
USB led does not light up and the device is not discoverable by Windows or MacOS. 
It is not shown in the Windows 10 Device Manager, or in the MacOS System Report, or found via `ioreg -p IOUSB` on MacOS.

It is also not discoverable by `hackrf_info` on MacOS, or any Mayhem firmware command line util on windows (including `dfu-util-static.exe` when set in DFU mode).

### Version information
**Operating system**: MacOS 11.6.1 and Windows 10 64bit.

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.
```

straithe, 2022-02-01:
Can you please post pictures of your HackRF (front and back) and Portapack here?

goa, 2022-02-01:
@straithe thanks for your quick reply. Here it is:
[image]
[image]
[image]
[image]

straithe, 2022-02-01:
I'm going to repeat back to see if I understand what you are looking for help with: You are trying to connect your HackRF+Portapack to the computer by USB, but when you plug it in the USB light does not come on and the device does not show up on your computer as connected, is that correct? Did the portapack battery have charge or was it drained?

Also, please note, it looks like both your HackRF and Portapack H1s are clones. That makes it harder to diagnose what may be happening as the designs of these pieces of hardware may not match what is here in the HackRF repository or the Portapack repository.

goa, 2022-02-01:
That is exactly right.

I have never installed a portapack battery, but I never had any issues
before.

After doing some research for my issue today, I also came to the conclusion
that my HackRF is a clone :(

I understand that if the designs are not exactly the same, we might not be
able to tell what’s wrong. Thank you very much for helping out.

straithe, 2022-02-02:
I'm confused. I wasn't asking if you installed a battery. The Portapack should have came with a battery installed, that's what enables you to power on the HackRF and use it with a Portapack when it isn't plugged into a computer. What I'm really interested in knowing is whether the Portapack had any charge or not when you plugged it and the HackRF into the computer. Do you know?

To get USB capability back, I suggest following the DFU steps in our documentation (https://www.health.gov.on.ca/en/public/publications/ohip/temp_foreign.aspx) while the Portapack is removed from the HackRF. Have you attempted that already?

miek, 2022-02-02:
> The Portapack should have came with a battery installed, that's what enables you to power on the HackRF and use it with a Portapack when it isn't plugged into a computer

The built-in battery is rarer on the H1s, the original PortaPack H1 doesn't have one and most of the clones don't - they were designed to be used with a USB battery pack.
Some clones added it it on, and I think all of the H2s have one.

straithe, 2022-02-02:
Ahhh! Good to know. Thank you for the correction. I appreciate it.

goa, 2022-02-02:
What @miek describes is exactly the case with my PortaPack H1. It didn't come with a battery.

Regarding the DFU steps, I have put the HackRF in DFU mode (without the PortaPack connected) but it's still not discovered as any kind of USB device by MacOS or Windows 10.

Running `dfu-util-static -l` from the Mayhem PortaPack firmware doesn't show any devices.

Are you referring to this guide? https://hackrf.readthedocs.io/en/latest/updating_firmware.html#only-if-necessary-dfu-boot

I currently don't have access to a linux installation, but if you think there's a chance that the device will be discoverable there in DFU mode, I'll try it.

goa, 2022-02-02:
Just wanted to let you know about a couple of other things I tried today:

1) Bought a new USB cable - nothing changed.
2) Connected the HackRF to an older computer that only had USB 2 ports (because I read there could be some issues with USB 3)

The USB led does not light up when in Mayhem's HackRF mode, but I'm pretty sure it works because it lights up dimly when I hold down the reset button.

I've tried using dfu-util on Windows 10, but the device is never discovered, even though I've put it in DFU mode with multiple different USB cable / port / computer combinations. 

Finally, I'm downloading a Pentoo image to check if I have more luck on Linux, but I doubt it :(

goa, 2022-02-03:
Just tried with Pentoo Linux, again HackRF is not detected by `hackrf_info` (in Mayhem's HackRF mode) or `dfu-util` (in DFU mode, without the PortaPack).

Furthermore, I don't see the device in the `dmesg` output, in either mode. Tried with two different USB cables, one of which was working perfectly with the device 3 days ago.

Noting again that the device itself works great with PortaPack and Mayhem, powered by the USB port.

goa, 2022-02-04:
After checking a few previous (and currently closed) GitHub issues, I physically examined U15 via a magnifying glass. It shows no physical damage.

straithe, 2022-02-09:
Thank you for all of the updates! At this point is sounds like a hardware fault to me. If you want to keep trying to fix it yourself, I suggest examining the USB connector or the LPC's USB peripheral. If you are done trying to fix this clone harware yourself, I suggest you try contacting your reseller to see if they will do an exchange.

goa, 2022-02-09:
Thanks for your help @straithe :)
I will have to agree. I'm not sure I can get a replacement after so long, so I'll do my best to try and fix it.
I see that other people have been having similar issues with clones, so perhaps I'll have to hone my hardware skills a bit.

Could you help me with one thing? What is the LPC's USB peripheral? 
I see in previous (closed) issues, that perhaps U15 could be damaged (instead of the LPC). Is that the peripheral you are referring to?

straithe, 2022-02-24:
The LPC4320 is the microcontroller on the HackRF. The LPC's USB peripheral is a part of that chip, so it is in that chip. 

If you would like to try replacing the U15 you are certainly welcome to! U15 is a protective piece. You can try de-soldering the part and testing the HackRF to see if that works. If it does work, you will probably want to replace U15 with a new part to get that protection back.

goa, 2022-02-24:
Thank you @straithe! I'll try that and see how it goes.
Do I understand correctly that the U15 is ESD protection for the LPC's USB peripheral?

straithe, 2022-02-24:
U15 is ESD protection.

closed 2022-02-24 by straithe