greatscottgadgets/hackrf #1044: Osmocom source not working with Gnuradio 3.10.1.1

K4prc, 2022-02-07:
### Steps to reproduce
1.fresh install of Ubuntu 20.04 and Gnuradio 
2.  open GRC and add osmocom source, connected to a qt gui frequency sink
3. run the block diagram

### Expected behaviour
I expect to get an FFT window showing the signal from the sdr.  I have tried it with both Airspy and HackRF, same error

### Actual behaviour
I get this error within GRC:

Executing: /usr/bin/python3 -u /home/john/Desktop/top_block.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2021.03.1
Traceback (most recent call last):
  File "/home/john/Desktop/top_block.py", line 194, in <module>
    main()
  File "/home/john/Desktop/top_block.py", line 172, in main
    tb = top_block_cls()
  File "/home/john/Desktop/top_block.py", line 132, in __init__
    self.connect((self.osmosdr_source_1, 0), (self.blocks_multiply_xx_0, 0))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 39, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 106, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7fae11a86cb0>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fae07c2eb40>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7fae0f312bb0>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.

>>> Done (return code 1)

>>> Done (return code 1)

### Version information
**Operating system**:*
Ubuntu 20.04.3

Hackrf_info output:
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82513735f
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x006a475e

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
Gnuradio 3.10.1.1

### Output
```
Insert any commandline or build output here

I found a couple of posts that indicated the problem could be due to the Osmocom source being set up for a earlier version of Gnuradio, so I tried rebuilding the source.  I clone gr-osmosdr, but when I run the cmake ../ command, I get an error (below).
Any assistance will be most appreciated.

John

-- Build type not specified: defaulting to release.
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:13 (find_dependency)
  CMakeLists.txt:45 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/john/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
-- Build type not specified: defaulting to release.
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:13 (find_dependency)
  CMakeLists.txt:45 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/john/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".

schneider42, 2022-02-07:
You are missing the `libspdlog-dev` package

K4prc, 2022-02-08:
Thank you, that eliminated my cmake error.  Unfortunately it just moved me to the next error, which I also do not know how to decipher.  Any idea why I would be running into these errors? 

ohn@ASUS-GL552VW:~/gr-osmosdr/build$ sudo make
[sudo] password for john: 
[  3%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/fcd/fcd_source_c.cc.o
/home/john/gr-osmosdr/lib/fcd/fcd_source_c.cc: In constructor ‘fcd_source_c::fcd_source_c(const string&)’:
/home/john/gr-osmosdr/lib/fcd/fcd_source_c.cc:147:36: error: no matching function for call to ‘fcd_source_c::connect(gr::fcdproplus::fcd::sptr&, int, gr::hier_block2::opaque_self, int)’
  147 |     connect( _src_v1, 0, self(), 0 );
      |                                    ^
In file included from /home/john/gr-osmosdr/lib/fcd/fcd_source_c.h:23,
                 from /home/john/gr-osmosdr/lib/fcd/fcd_source_c.cc:29:
/usr/include/gnuradio/hier_block2.h:83:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr)’
   83 |     void connect(basic_block_sptr block);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:83:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:93:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:93:35: note:   no known conversion for argument 1 from ‘gr::fcdproplus::fcd::sptr’ {aka ‘boost::shared_ptr<gr::fcdproplus::fcd>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                  ~~~~~~~~~~~~~~~~~^~~
/home/john/gr-osmosdr/lib/fcd/fcd_source_c.cc:156:36: error: no matching function for call to ‘fcd_source_c::connect(gr::fcdproplus::fcdproplus::sptr&, int, gr::hier_block2::opaque_self, int)’
  156 |     connect( _src_v2, 0, self(), 0 );
      |                                    ^
In file included from /home/john/gr-osmosdr/lib/fcd/fcd_source_c.h:23,
                 from /home/john/gr-osmosdr/lib/fcd/fcd_source_c.cc:29:
/usr/include/gnuradio/hier_block2.h:83:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr)’
   83 |     void connect(basic_block_sptr block);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:83:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:93:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:93:35: note:   no known conversion for argument 1 from ‘gr::fcdproplus::fcdproplus::sptr’ {aka ‘boost::shared_ptr<gr::fcdproplus::fcdproplus>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                  ~~~~~~~~~~~~~~~~~^~~
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:128: lib/CMakeFiles/gnuradio-osmosdr.dir/fcd/fcd_source_c.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:355: lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

straithe, 2022-02-24:
I suspect you might have multiple version of GNU Radio or gr-osmodr being installed on your computer. I recommend removing all versions of GNU Radio and gr-osmodr and starting again.

straithe, 2022-03-21:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue, respond here, or open a new issue if you still need assistance.

closed 2022-03-21 by straithe

dizcza, 2022-06-09:
I'm having the same issue.
gnuradio 3.10.2.0 installed as the wiki suggests - from PPA.
gr-osmosdr 0.2.0-2

```
Executing: /usr/bin/python3 -u /home/dizcza/Projects/hackrf/top_block.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:1,2), Device or resource busy.

Using HackRF One with firmware git-dfadf6a
Traceback (most recent call last):
  File "/home/dizcza/Projects/hackrf/top_block.py", line 201, in <module>
    main()
  File "/home/dizcza/Projects/hackrf/top_block.py", line 179, in main
    tb = top_block_cls()
  File "/home/dizcza/Projects/hackrf/top_block.py", line 142, in __init__
    self.connect((self.osmosdr_source_0, 0), (self.qtgui_freq_sink_x_0, 0))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 39, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 106, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7ff54437ef70>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7ff541e78a50>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7ff54627e8b0>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.

>>> Done (return code 1)
```

So sad. I have bought a HackRF, it's lying beside me and I cannot even execute the first lesson. Such dreadful user experience with gnuradio and osmosdr.

Not sure though if the issue should be posted here - I think it's purely a gnuradio and osmosdr problem, isn't? I've googled the problem and found this issue here.

dizcza, 2022-06-09:
It's indeed strange that the console output shows `gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0` although I've installed gnuradio of another version:

```
$ gnuradio-config-info -v
3.10.2.0
```

BrightBeacon, 2025-09-30:
Just work for me：
$ conda install gnuradio gnuradio-osmosdr
$ gnuradio-companion