greatscottgadgets/hackrf #1046: Osmocom Source return -11 on gnuradio 3.10.1.1

Caglarr, 2022-02-09:
Hi Dear,

I'm currently trying to execute NFM Receiver project and I used osmocom source with hackrf one for the purpose of receiver. But I couldn't execute. 
### Steps to reproduce
1. I installed gnuradio 
2. Then pull hackrf git repo and make with cmake
3. Then I create the project

### Expected behaviour
I was expecting the project I created to run and generate a graphical output with Qt Gui Sink.### Actual behaviour
Tell us what happens instead
### Actual behaviour
The situation that occurs is that no graphics and sound output occur. and return code -11 is returned.
[image]

### Version information
Ubuntu 20.04
Gnu Radio Companion 3.10.1.1 (Python3.8.10)

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
Gnu Radio Companion 3.10.1.1 (Python3.8.10)
### Output
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.0.0
built-in source types: file rtl_tcp rfspace soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] Using format CF32.

>>> Done (return code -11)
Insert any commandline or build output here
```
Thanks in advance,
Best regards..

Caglarr, 2022-02-10:
Hi again,
May be this information give a hint for solution. When I tried run "osmocom_fft -a hackrf -v" command from terminal, output was like this,

> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.0.0
> built-in source types: file rtl_tcp rfspace soapy redpitaya 
> [INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
> [INFO] Using format CF32.
> Supported sample rates 48000-48000 step 0.
> Using auto-calculated mid-point frequency
> Supported frequencies 0.0-6.0G
> Traceback (most recent call last):
>   File "/usr/local/bin/osmocom_fft", line 714, in <module>
>     main()
>   File "/usr/local/bin/osmocom_fft", line 690, in main
>     tb = app_top_block(qapp.arguments(), "osmocom Spectrum Browser")
>   File "/usr/local/bin/osmocom_fft", line 297, in __init__
>     self.scope_win = sip.wrapinstance(self.scope.pyqwidget(), Qt.QWidget)
> **AttributeError: 'gnuradio.qtgui.qtgui_python.freq_sink_c' object has no attribute 'pyqwidget'**
> **Segmentation fault (core dumped)**

Best regards.

schneider42, 2022-02-13:
See https://github.com/gnuradio/gnuradio/issues/5175

closed 2022-04-29 by straithe