greatscottgadgets/hackrf #1047: hackrf_sweep Output File not Appearing, -r Flag not Valid

aldenkane, 2022-02-10:
### Steps to reproduce
1. sudo apt update
2. sudo apt install libhackrf0
3. sudo apt install hackrf
4. hackrf_sweep -f 2200:2400 -r sample_output

### Expected behaviour
After running hackrf_sweep, I would either expect a default output file when no -r flag is provided, or an output file in my pwd when the output filename is provided with the -r flag. From the docs, "running hackrf_sweep -f 2400:2490 gives the following example results: ..." and then an output table is shown in the docs. I am not finding a way to get a similar table of data to analyze the sweep.

### Actual behaviour

I see `hackrf_sweep: invalid option -- 'r'` when running the command with the -r flag. I cannot locate an output file when running the command without the -r flag.

### Version information
**Operating system**:
Raspbian GNU/Linux 10 (buster)

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc286280c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x005d4358
```

### Output
Output from step 3 to reproduce:

```
pi@alden:~/test $ hackrf_sweep -f 2200:2400 -r sample_output
hackrf_sweep: invalid option -- 'r'
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 16384-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-B] # binary output

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .
pi@alden:~/test $ 
```

When I run with no -r flag:

```
pi@alden:~/test $ hackrf_sweep -f 2200:2400
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2200 MHz to 2400 MHz
Stop with Ctrl-C
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
^CCaught signal 2
35.9 MiB / 0.897 sec = 40.1 MiB/second

Exiting...
Total time: 3.89694 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

### Other Comments

Perhaps this issue comes from the libhackrf api being mismatched with the firmware version. The firmware is up to date, and I freshly installed / reinstalled hackrf packages in troubleshooting. I couldn't find any information on ensuring compatibility here, and hackrf_info was returning unknown versions for hackrf and libhackrf.

I power cycled both the Raspberry Pi and HackRF in my troubleshooting attempts.

The somewhat related threads I found on other forums didn't give definite resolution steps.

Thank you in advance for the help.

gozu42, 2022-02-10:
afair raspian is shipping a 2018 version of hackrf lib/tools.
you can check what version you have installed like so: dpkg -l | grep hackrf

recommend you build/install either the current release or even git version.

https://github.com/greatscottgadgets/hackrf/releases

aldenkane, 2022-02-18:
@gozu42 Thanks for the exceptional help. Installing hackrf lib/tools from 2021.03.01 release at the link you provided, instead of from apt install, resolved this issue. I can now run `hackrf_sweep` fine.