greatscottgadgets/hackrf #1048: possible hardware defect

ecc1, 2022-02-13:
I have 2 HackRF Ones. I've been using one for years, and it works fine. The other was still in a sealed box until today, when I wanted to upgrade the firmware on both units. The second unit seems to be defective.

### Steps to reproduce
Use gqrx to listen to a local FM station.

### Expected behaviour
FM reception should continue indefinitely.

### Actual behaviour
Reception stops after approximately 5 minutes (sooner if changing frequency), with error messages like
"Failed to stop RX streaming (-1000)" and
"hackrf_set_freq(4.51291e+08) has failed (-1000) No such device (it may have been disconnected)".

### Version information
Debian testing
hackrf and libhackrf0 packages version  2021.03.1-2

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8389869cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x006a435e

Photos of the suspect unit are attached.

[image]
[image]

martinling, 2022-02-13:
You say you have one unit failing and the other working - have you tried them both on the exact same USB cable plugged into the same port? It sounds like it's just disappearing off the USB bus completely, so I'd just like to rule out any other hardware factors.

ecc1, 2022-02-13:
Yes, I'm just switching the micro end of the USB cable from one unit to the other.

martinling, 2022-02-13:
It may be worth checking again with current git master versions of the firmware & host packages, as there are a few intermittent bugs caused by race conditions that we've fixed since the 2021.03.01 release.

In particular, we had some problems that were showing up as a loss of reception when changing parameters in gqrx (#883) and those were fixed with the changes in PR #1029.

ecc1, 2022-02-13:
But is there enough variability between hardware builds that this problem could occur consistently with one unit and never with another?

straithe, 2022-02-14:
No. That is abnormal.

martinling, 2022-02-14:
For firmware problems that have a race condition involved it's plausible for one unit to be more prone than another, e.g. due to slight differences in oscillator frequencies.

I'm certainly not ruling out a hardware fault here - but it does seem odd to have a fault that shows up only after several minutes _and_ happens more quickly if making changes in gqrx. It's hard to see what failure mechanism would give that sort of result, whereas I just fixed several race conditions with very similar symptoms.

So it'd be interesting to know if you get the same result on current master - or for that matter the previous 2018.01.1 release, since that release seems to have been less prone to timing bugs. If the behaviour is the same with other firmware versions, then it seems like it must be a hardware fault, albeit a strange one. Whereas if it works with some different version, then that may indicate there is yet another race condition I need to track down and fix in the firmware.

So I would much rather find it's a hardware fault! We have a simple solution for those: you contact your reseller for a return.

roadmutt, 2022-02-14:
I need to re solder a tiny little resistor on one of my hacked sdrs

ecc1, 2022-02-14:
I built and flashed the current firmware from git, but the same problem occurs:

> Failed to stop RX streaming (-1000)
> Failed to close HackRF (-1000) No such device (it may have been disconnected)

It doesn't always take 5 minutes to fail; sometimes it happens almost immediately (that was the case for the released version too).

martinling, 2022-02-14:
OK, thank you for the additional testing to confirm.

It seems pretty clear that this is a hardware fault. You can contact your reseller to arrange a return.

closed 2022-02-24 by straithe