greatscottgadgets/hackrf #1049: i put 19v in the rf :(

andreasr81, 2022-02-14:
hi, i make a dump mistak yesterday, i switched the sat cable that providing 19v for the lnb and put it in my hackrf, the hackrf is still recognicing by the pc for a moment, and is also receiving,  but near the usb plug its getting very hot,  and the usb light beginn to dimming i think there is a shortage somewhere, maybe someone had a simular problem and no what to do?

martinling, 2022-02-14:
Hmm. I'm impressed that it's still working at all after getting 19V on the RF input, since that would lead to current going through the body diode of Q4 into VAA, and from there through Q3 into VCC. I would have expected widespread damage to parts all over the board.

But perhaps that route has also somehow saved you, if something downstream managed to effectively clamp the input to a lower level, burning itself out in the process. So it might be possible to repair if you want to do so.

If you can identify exactly which component is getting hot, then if you're lucky, replacing that component might be sufficient. Close to the USB port, my guesses would be either the regulator U21 in the corner, or the TVS array U15.

In fact U15 seems quite likely, since the current path I described above might continue backwards through the regulator and the body diode of Q5, eventually getting clamped to ~6.5V by U15. For a brief period, that might make the event survivable.

If U15 is the problem, you can actually remove it completely for testing purposes. Its function is to protect the USB VBUS and data lines from transient voltage events. So the HackRF will be vulnerable to static discharge at the USB port without it, but it will still connect to a host.

andreasr81, 2022-02-15:
thanks a lot for giving me that informations, i have to ask a friend to help me there.

closed 2022-03-17 by martinling

helmet21, 2022-04-12:
Hi!

Sorry to post in a closed issue but I tried to access the discord to send my question but was unable due to problems with the registration process. Also I thought it might be better to post in this issue since my question is very similar.

I also suspect I fed voltage into the RF input and the result is a component becoming very hot. The HackRF seems to turn on normally for a second or two but then the LEDs start flickering and dim.

I checked the components and there is one becoming very hot (almost instantly goes to 47 degrees celsius). The text on the component says "CAT OCI AVHR" but I can't find any information about it online and I can't find it on the schematics.

So I was wondering if maybe someone could tell me which component this is and how I can find a replacement.

Best regards
helmet21

[image]

straithe, 2022-04-12:
Please open a new GitHub issue and we'll be happy to help you out.