greatscottgadgets/hackrf #1050: Unable to transmit using hackrf one and receive on external ham radio

jeffLunar, 2022-02-15:
### Steps to reproduce
1. Download gnuradio through the Radioconda installer
2. Create the following in GNU radio companion
[image]

### Expected behavior
The .wav file should be broadcasted to the desired frequency and a ham radio should be able to receive it.

### Actual behavior
Ham radio stays silent. Unable to hear anything from the Hackrf One.

I am fairly new to this so I could simply be overlooking something that is causing no output or possibly even receiving it incorrectly.

### Version information
Windows 11
GNU Radio Companion - v3.8.2.0-57-gd71cd177 (Python 3.9.0)

**hackrf_info output:**
GNU Radio Companion - v3.8.2.0-57-gd71cd177 (Python 3.9.0)

### Output
```
Generating: 'C:\\Users\\lunar\\Documents\\gnuradio\\transmit.py'

Executing: C:\Program Files\GNURadio-3.8\tools\python3\python.exe -u C:\Users\lunar\Documents\gnuradio\transmit.py

len(interp_taps) = 244
gr::pagesize: no info; setting pagesize = 4096
```

Thanks in advance!

dmaltsiniotis, 2022-02-15:
Freeeeeebiirrrrrrrrd!

This looks like an issue with GNU Radio specifically rather than with the HackRF One or its supporting tools and libraries. Since this issue section pertains to the usage of just the HackRF One device, I'm not sure I can be of much help as I'm not well-versed in GNU Radio.

However, I would probably start off by validating that the HackRF One is operating correctly, independently of GNU Radio. This can be validated by opening a command prompt (in this case I would probably recommend using the prompt that comes with radioconda under  Start->radioconda->Conda Prompt), and issuing the command: `hackrf_info`

You should see an output similar to this:
[image]

Also, just trying to eyeball the flowgraph, it looks like you have the sample rate of the osmocom sink set to 48k, but the HackRF One has a minimum sample rate of 2Msps, could this be related to the issue perhaps?

jeffLunar, 2022-02-15:
Thanks for the quick reply!

I was able to validate the hackrf by using the command `hackrf_info` which is shown below:
'''
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc3273a707
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0061435b
'''

Regarding the sample rate, increasing the sample rate above the 2Msps rate did not change the behavior of the device. If you believe this is an issue regarding gnuradio rather then hackrf, I can take a look over there. 

I was wondering if there are other preferred software packages to use with hackrf other than gnu radio companion.

dmaltsiniotis, 2022-02-16:
Hi @jeffLunar ,

From your output, it appears as if the HackRF device is properly configured and responding, and it has the most recent version of the firmware installed as well, which is nice. This leads me more toward thinking the GNU Radio flowgraph needs to be tweaked.

From some cursory research I came across others espousing the importance of a > 2Msps when transmitting (https://factorialabs.com/2017/12/14/hackrf-ones-minimum-sample-rate-spec-there-for-a-reason/) so I think that's probably a place to start. However, and I'm guessing here, I would assume the signal stream needs to be modified from the 48k (or 44.1k) samples per second the wav file provides, to match the transmit sample rate of the device, whatever that is (in this case targeting 2Msps). Here is an example flow graph I found online of someone doing just that: https://www.dlineradio.co.uk/articles/building-a-very-low-power-fm-transmitter-with-hackrf-one/

Regarding other preferred software packages, that depends entirely on what your goals are and what you wish to accomplish.

GNU Radio is sort of the low-level "Swiss Army Knife" of SDR, providing a highly customizable RF/signal processing framework. However it can be unruly for first-time users.

For receive only: If you'd like to do some basic frequency spectrum scanning/listening, there are a few of really good (free) Windows apps that come with basic de-modulators build-in such as:  AM, wide-band FM, narrow-band FM, single and double side-band and more:
* SDR # - The most popular water-fall style RF scanner. Lots of plugins available: https://airspy.com/download/
* SDR++ - A newcomer to the SDR scene, very fast/lightweight. "modules" based: https://github.com/AlexandreRouma/SDRPlusPlus/releases
* CubicSDR - Been around for a while, but totally cross platform and supports quickly adding multiple de-modulators at once for simultaneous listening: https://cubicsdr.com/?cat=4
* GQRX - Already included with your radioconda installation, just launch it from the conda terminal: https://gqrx.dk/download
* HDSDR - Another windows-only option that some folks like: https://www.hdsdr.de/
* ...and many more that have more specialized use-cases.

For digital signal decoding and analysis, Universal Radio Hacker (URH) is my go-to. It has the ability to record directly from SDR's but it's bread and butter is not in the waterfall/recording, it's in processing signals and turning the waves into bits and bytes for analysis: https://github.com/jopohl/urh/releases

For transmit based goals, going the GNU Radio route is probably the best if you're already there. You can customize your transmission to your needs exactly.

Anyway, I hope this helps point things in the right direction.

Cheers,

-DM

jeffLunar, 2022-02-24:
@dmaltsiniotis 

The 2 Mbps was the inherent problem that resulted in the failed transmission.

closed 2022-02-24 by jeffLunar