greatscottgadgets/hackrf #1051: TX Power amplifier (U25) damaged

colisee, 2022-02-17:
### Context
I purchased the HackRF One card and the ANT500 antenna on November 11th last year and everything was working properly until, for no obvious reason, the TX power amplifier (U25) stopped working a few weeks ago. 
For your information, I never connected the output of the HackRF to anything else than the ANT500 and the only thing that I did was to transmit at full power via the SoapySDR interface for the HackRF (TXgain=63) on channel 5A of the European DAB band (174.928Mhz).

### Test conditions:
- Software: ODR-DabMux + ODR-DabMod + SoapySDR
- Hardware: Sony XDR-P1DBP

### Steps to reproduce
1. TXGAIN=23, signal strength [43 - 50]
2. TXGAIN=24, signal strength 0
3. TXGAIN=37, signal strength 0
4. TXGAIN=61, signal strength [40 - 45]

### Expected behaviour
TXGAIN=23 is equivalent to hackrf_transfer -x 23 -a 0
TXGAIN=24 is equivalent to hackrf_transfer -x 10 -a 0
TXGAIN=37 is equivalent to hackrf_transfer -x 23 -a 0
TXGAIN=61 is equivalent to hackrf_transfer -x 47 -a 1

I would expect the signal strength to increase with a higher TXGAIN (this is what was happening at the beginning)

### Actual behaviour
See above

### Version information
**Operating system**: Linux Debian Bullseye

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc263e94c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0054436b

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

martinling, 2022-02-21:
Hi @colisee,

A TX amplifier failure would mean you get no signal with the amplifier enabled, but that doesn't match the results you've described. You say you're getting zero signal at TXGAIN 24 and 37, but you've also said that those correspond to a `hackrf_transfer` setting of `-a 0`, which means the amplifier is off. In the case of TXGAIN 61 where the amplifier is on (`-a 1), you're getting a signal.

A TX amplifier failure would mean no signal when `-a 1`, but that isn't the pattern shown here.

If you think you have a TX ampfailure it would be best to confirm this with a simpler test, e.g. just using `hackrf_transfer -c` to send a CW signal with `-a 0`, picking that up on some receiver, and confirming whether it disappears if you change to `-a 1`.

With all the DAB parts involved there are a lot of other things that could be going wrong.

colisee, 2022-02-22:
Hi @martinling ,

Thank you very much for your reply. Further to your suggestion, I changed the test procedure. I took a SONY ICF-2010 receiver which features a signal-strength indicator (10 leds), I switched the frequency to 136 Mhz and I am now using the following command:
`sudo hackrf_transfer -f 136000000 -c 100 -n 50000000 -x TX_gain -a RF_gain`

Here are the signal strength readings with different values of _TX_gain_ and _RF_gain_:

TX gain | RF gain | Signal strength
:---:|:---:|:---:
10 | 0 | 5 / 10
23 | 0 | 7 / 10
47 | 0 | 10 / 10
10 | 1 | 0 / 10
23 | 1 | 0 / 10
47 | 1 | 6 / 10

As you can see, the results are quite similar to those obtained with the ODR-mmbTools (DAB) suite. 
Looking at other posts, this situation seems typical to the U25 being damaged...
Thus, **what did I do** to damage U25, knowing that the HackRF was connected to its ANT500 all the time?

Kind regards.

martinling, 2022-02-22:
Thanks @colisee, that's a very good test and does look like a TX amplifier failure.

I think you must have meant to write `-a 1` in the second and third lines here - that's why the previous result was confusing:
```
TXGAIN=23 is equivalent to hackrf_transfer -x 23 -a 0
TXGAIN=24 is equivalent to hackrf_transfer -x 10 -a 0
TXGAIN=37 is equivalent to hackrf_transfer -x 23 -a 0
TXGAIN=61 is equivalent to hackrf_transfer -x 47 -a 1
```
You may not have done anything in particular to cause the failure - I don't see anything wrong with the usage you've described. It's possible it was just a bad part that's taken a while to fail so didn't show up in factory testing. You should be able to arrange a return & replacement with your reseller.

colisee, 2022-02-22:
Thank you @martinling for proposing a return/replacement procedure. I will close this issue. 

Regarding my first tests, you are totally right: with soapysdr for the HackRF: a txgain greater than 23 implies that the RF amplifier is activated. I don’t totally agree with the algorithm  implemented for soapysdr, since it doesn’t offer all the combinations you get with hackrf_transfer and I will try to see if it could be changed. 

Kind regards.

closed 2022-02-22 by colisee