greatscottgadgets/hackrf #1053: [FEATURE REQUEST] Please change the USB connector to a Type-C port

3Domse3, 2022-02-23:
### Steps to reproduce
1. Buy a HackRF One in 2022
2. Get annoyed that USB-C is not supported in 2022
3. Post a Github issue since you obviously have nothing better to do

### Expected behaviour
Accept USB-C

### Actual behaviour
Not accepting USB-C

### Version information
**Operating system**: every

**hackrf_info output:** none

R7c0, 2022-02-23:
USB C would be great in a future revision.

straithe, 2022-02-24:
Thank you for your suggestion! I also really appreciate thank you filled out the bug report. We are not looking to add USB-C support to HackRF One. If we do make another SDR product in the future, we will take USB-C support into consideration.

closed 2022-02-24 by straithe