greatscottgadgets/hackrf #1054: hackrf one : hackrf_info version: git-e93d70e , libhackrf version: unknown (0.5)

KimlongSeng, 2022-02-23:
### Steps to reproduce
1. run : $ sudo apt-get install hackrf
2. run : $ hackrf_info
3.

### Expected behaviour
Tell us what you expect should happen

hackrf_info version: 2018.01.1-2 ?
libhackrf version: 2018.01.1-2 ?
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc317b3f07
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00744357

### Actual behaviour
Tell us what happens instead

hackrf_info version: git-e93d70e
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc317b3f07
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00744357

### Version information
**Operating system**: Raspberry Pi 4 (Linux)

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GNU Radio Companion 3.7.14.0
### Output
```
Insert any commandline or build output here

pi@pisdr:~ $ sudo apt-get install hackrf

Reading package lists... Done
Building dependency tree       
Reading state information... Done
hackrf is already the newest version (2018.01.1-2).
The following packages were automatically installed and are no longer required:
  liba52-0.7.4 libaribb24-0 libbasicusageenvironment1 libcddb2 libdav1d3 libdvbpsi10 libebml4v5 libgroupsock8 libixml10
  liblirc-client0 liblivemedia64 liblua5.2-0 libmatroska6v5 libmpeg2-4 libopenmpt-modplug1 libplacebo7
  libprotobuf-lite17 libproxy-tools libqt5x11extras5 libresid-builder0c2a libsidplay2 libspatialaudio0 libupnp13
  libusageenvironment3 libvlc-bin libvlc5 libvlccore9 libxcb-xv0 vlc-bin vlc-data vlc-l10n vlc-plugin-notify
  vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2 vlc-plugin-video-output vlc-plugin-video-splitter
  vlc-plugin-visualization
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```

straithe, 2022-02-24:
I'm not quite certain what your concern is. Can you please provide more details?

KimlongSeng, 2022-02-24:
hackrf_info version: git-e93d70e
libhackrf version: unknown (0.5)

I install the libhackrf, but it just reported as unknown. 

Thank you for taking the time to respond.

straithe, 2022-03-21:
In many scenarios, having the libhackrf version be unknown is normal. As for your hackrf_info version being different, how did you install the HackRF software?

KimlongSeng, 2022-03-23:
Just using 
$ sudo apt-get install hackrf

Like you said, the hackrf one is working fine with the version and libhackrf = unknown(0.5).

I was just curious.

closed 2022-03-23 by KimlongSeng