greatscottgadgets/hackrf #1055: hackRF one RF led red

omurugur, 2022-02-26:
I have a hackrf one device in my hand and when I connect this device to the computer, I cannot receive broadcasts and the rf lamp is red and the 1v8 lamp is orange. I updated the firmware to the latest version but the problem did not go away. I plug in the pineapple antenna but I can't get any radio.

straithe, 2022-03-08:
Can you please describe your set up, such as what software you are using, and how you are going about testing your HackRF?

omurugur, 2022-03-09:
I updated to the latest version, I connected the pineapple antenna to my device. universal radio hacker , HDSDR ,gnu radio does not work on any of the installed windows. red led is on in the rf area

miek, 2022-03-09:
Just to clarify: the colours of the LEDs don't mean anything, so a red LED being lit up doesn't mean that something is wrong. See the FAQ for more detail: https://hackrf.readthedocs.io/en/latest/faq.html#why-are-the-leds-on-hackrf-different-colours

straithe, 2022-03-21:
@omurugur are you still having any concerns, or are you happy with the explanation of the LED colours?

closed 2022-04-29 by straithe