greatscottgadgets/hackrf #1059: Question on transmission gains

colisee, 2022-03-06:
Hi,

please excuse me in advance if I am asking a stupid question, for I am not an expert in electronics.

Is there a difference in terms of transmitted signal power between the following 2 commands:
hackrf_transfer -x 44 -a 0 …
hackrf_transfer -x 30 -a 1 …

If the result of both commands is equivalent in terms of transmitted power, do you have any recommendations?

Thank you for your help.

martinling, 2022-03-06:
I would not expect those to be equal.

In the [datasheet](https://www.farnell.com/datasheets/7246.pdf), the gain of that RF amplifier varies over around 10-12dB according to frequency.

[image]

In practice the gain may be less due to compression (if operating near maximum output power), as well as mismatch on the output and other factors.

Unfortunately, the gain of this amplifier is commonly quoted as being 14**dB**, most likely due to common misunderstandings of the datasheet, which gives a headline output power rating of 14**dBm**. (The two are not equivalent - the former is a relative measurement and the latter is absolute. You can have a 14dBm amplifier with only 4dB of gain, so long as you feed it with 10dBm input).

When preparing the current [block diagram](https://hackrf.readthedocs.io/en/latest/_images/block-diagram.png) I gave a figure of ~11dB for the sake of brevity, but I would expect this to reduce as compression kicks in, if the signal from the previous stage is large enough.

It would be good to correct that 14dB figure, but it has found its way to many places, including UI controls thanks to the ranges presented by e.g. [this line in the osmosdr code](https://github.com/osmocom/gr-osmosdr/blob/master/lib/hackrf/hackrf_source_c.cc#L318), and similar elsewhere.

colisee, 2022-03-06:
Hi @martinling 

thank you very much for your detailed answer. I understand that the gain of the amplifier varies depends on the broadcasted frequency. 

Let’s assume, for simplicity sake, that the amplifier gain was 14db. In that case, would the 2 commands I mentioned be equivalent. 

- If yes, would you recommend one option against the other?
- If not, do you have an explanation « for dummies » ?

Kind regards.

martinling, 2022-03-06:
It's best not to run any amplifier near its limits unless you really have to. They are happiest in the middle of their range.

No amplifier is perfectly linear - they cannot maintain a constant gain as the output demands increase. As they work harder, the peaks start to be less tall than they should be relative to the rest of the signal. This is what is meant by compression. Even long before the peaks start visibly clipping, that subtle distortion starts to show up in the frequency domain, appearing as increased harmonics emitted at odd multiples of the operating frequency, and as a reduction in the clarity of the intended signal.

So run any individual amplifier gently, for cleanest results. That also means that if you need a fixed amount of total gain, it's usually better to spread that out across multiple amplifiers, rather than making one do all the work.result.

Which means that in your scenario, I would go with the 30dB + 14dB option.

colisee, 2022-03-07:
Crystal clear!
Thank you very much @martinling for your time

closed 2022-03-07 by colisee