greatscottgadgets/hackrf #1061: Fast switching between TX and RX

Jcwscience, 2022-03-14:
How quickly can the device switch between TX and RX? I don’t know of software right now that can switch quickly, but what is the theoretical limit? If an sdr program had enough control could an instant PTT be done?

I apologize if this is covered somewhere else but I’m a bit new to the platform.

Thanks!

martinling, 2022-03-14:
See #745 for some more recent discussion on this. 

The fundamental limit is defined by how long the `rf_path_set_direction()` function in the firmware takes to execute. That has to send some commands over serial buses to the MAX2837 and MAX5864 chips, so it can't happen instantly, but it could be sub-1ms.

In the last comment on that previous discussion, @mossmann noted that to hit that figure, timed operations in the firmware would need to be implemented (which is the very long standing issue #86).

In fact there has been some progress in that direction lately, because I added the basic firmware support for timed operations as part of PR #982, merged just recently.

At the moment though, there is no way for the user to request timed operations. We still need to figure out what the interface for that will looks like at several layers: the interactions between the M4 & M0 cores, the USB protocol level, the libhackrf API level, and the user-facing command line tools.

So the groundwork is there if someone's keen to work on this, but there isn't fast switching out of the box yet.

Jcwscience, 2022-03-15:
@martinling Thats good news that it’s at least possible! I was brainstorming possible interface methods and I thought a wide compatibility implementation might work like this: the driver registers a second dummy device as the transmit, and the SDR software could pick it up as a separate device, then when anything is being transmitted it can just return a blank IQ signal on the receive side so any programs with transmit support built-in already wouldn’t know the difference. Just an idea, I know practical implementation of these things is a nightmare.

martinling, 2022-03-15:
The interface for this is tricky, because in order for an "instant" RX->TX switch to happen, the host needs to have already transferred the samples it wants to transmit to the HackRF, before the HackRF stops RX. Otherwise, no matter how fast the HackRF itself can switch from RX to TX, there will be a gap during which it has no samples ready to transmit.

The problem is compounded by the fact that TX and RX both use the same 32KB buffer, so the TX data can't be written until the RX side is done with it. And we can't easily use a different buffer, because that particular 32KB has specific properties: it straddles a block boundary, such that each half has its own AHB bus connection, allowing the USB peripheral to read/write one side of it whilst the M0 core accesses the other, without bus contention.

There are tricks we could do, like writing the initial TX data to some other area of RAM, and having the M0 core read from there for the first samples, switching to the normal USB-filled buffer once it has exhausted them.

But that still leaves another issue: USB bandwidth. There is only just enough bandwidth on 480Mbps USB2.0 to keep up with HackRF's 20Msps max sample rate. And the upstream and downstream traffic have to share that bandwidth. So if you want to preload TX data whilst RX is still running, you won't be able to do that at full sample rate.

The other big problem is the round trip latency to the host. In most cases where someone's looking for timed operations and fast RX->TX switching, it's because they're dealing with a protocol which requires receiving some packet and then transmitting a response within a specified timeslot right afterwards.

But even if the HackRF is capable of that, it may not be possible in practice for the host to receive the RX samples over USB, decode them, generate a response, and get the TX samples back over USB to the HackRF in time for the transmit timeslot.

So what's possible in terms of fast RX->TX depends very much on the use case: what sample rate is needed, what turnaround time is needed, and what round trip latency can be achieved by the host. Designing the right interface probably needs to start with making a list of use cases that are actually achievable.

Jcwscience, 2022-03-15:
@martinling Ok, that makes sense. Would you happen to know where I can find more info on the way the flow of data is handled? I’d like to get a better grasp on the hardware

g7eix, 2022-03-15:
> For the hardware side of things, there is a block diagram.

Remember the good old days when we had a dedicated receiver and a dedicated transmitter in the shack - I suppose a solution would be to do the same with the HackRF. One dedicated to RX one to TX. I mean, we've all got more than one haven't we? We've all got the same addiction. :-)Just thinking to myself while I sip my coffee.  Richard W4/G7EIX

Jcwscience, 2022-03-15:
@g7eix Yeah but half the fun is rigging something to do what it wasn’t necessarily meant to do! At least that’s what I tell myself when I’m too broke to get the right equipment….

g7eix, 2022-03-15:
100% agree!

straithe, 2022-07-14:
I'm closing this issue as it seems to be resolved.

BW-SK, 2026-03-04:
Sorry for necro-posting, but this issue *isn't* resolved. Currently, there is no way to do this (especially not sample-accurate).

Buying *two* HackRF Pros for something that in theory should only require one is a bit of a surprise to us. I totally understand that for the HackRF One, the argument "just buy two" pulls a lot more weight.