greatscottgadgets/hackrf #1064: hackrf_transfer returns undefined symbol error

Jcwscience, 2022-03-15:
### Steps to reproduce
Run hackrf_transfer

### Expected behaviour
Transmit

### Actual behaviour
hackrf_transfer: symbol lookup error: hackrf_transfer: undefined symbol: hackrf_get_m0_state

### Version information
Ubuntu Impish aarch64

hackrf_info version: git-3344ea8
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc275297c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x005d4366

martinling, 2022-03-16:
`hackrf_get_m0_state` was just added to the library recently in #982. It looks like you're trying to run a new `hackrf_transfer` against an installed `libhackrf.so.0` that's not as up to date.

Jcwscience, 2022-03-16:
Ok that makes sense, I’ll hunt down the library and remove it and rebuild the project. Thanks!

Jcwscience, 2022-03-16:
@martinling That worked, I had a system installation still that I forgot about

closed 2022-03-16 by Jcwscience