greatscottgadgets/hackrf #1066: HackRF doesn't work

0l4y4, 2022-03-16:
### Steps to reproduce
1. I have a remote control that works in 868 MHz (I have tested with SDR device)
2. `hackrf_transfer -s 2000000 -f 868000000 -r test`
3. `hackrf_transfer -s 2000000 -f 868000000 -t test` or  `hackrf_transfer -s 2000000 -f 433000000 -t test -a 1 -x 24`
4.  It doesn't work and SDR does not receive the signal

### Expected behaviour
receive the signal on SDR

### Actual behaviour
Nothing

### Version information
Ubuntu 20.04 LTS

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2d1e495f
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x005c4756

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
call hackrf_set_sample_rate(2000000 Hz/2.000 MHz)
call hackrf_set_freq(868000000 Hz/868.000 MHz)
Stop with Ctrl-C
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 4.2 MiB / 1.000 sec =  4.2 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 4.2 MiB / 1.000 sec =  4.2 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 3.9 MiB / 1.000 sec =  3.9 MiB/second
 4.2 MiB / 1.000 sec =  4.2 MiB/second
 1.6 MiB / 1.000 sec =  1.6 MiB/second

Exiting... hackrf_is_streaming() result: streaming terminated (-1004)
Total time: 13.00420 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

straithe, 2022-03-21:
I'm a bit confused by what you've typed up and do need a bit of clarification. Are you saying that you cannot receive signals on your HackRF?

0l4y4, 2022-03-22:
@straithe 

I am listening with a SDR nooelec on another computer when I am sending signals with HackRF

martinling, 2022-03-22:
Hi @0l4y4,

From the commands you've given it looks like you're trying to replay a signal:
```
hackrf_transfer -s 2000000 -f 868000000 -r test
hackrf_transfer -s 2000000 -f 868000000 -t test
```
This isn't the right way to try to do that, for several reasons:
- `hackrf_transfer` isn't a replay tool. It's a low level utility that does not have the necessary capabilities to do this properly. You need to be able to see the signal you're capturing to do this correctly.
- You're not doing anything to control the receive gain. In order to receive a signal, it needs to be scaled to a level at which the baseband ADCs can capture it effectively.
- You're tuning the HackRF centre frequency to that of the signal you want to capture. But that frequency will be converted into the receiver's DC blind spot.  To capture a signal cleanly on a quadrature SDR, you need to tune at an offset from it.
- You're not doing any filtering to isolate the signal of interest. With that 20MHz sample rate, you would be indiscriminately retransmitting everything from 858 to 878 MHz.

I suggest checking out our [SDR video course](https://greatscottgadgets.com/sdr/). Replaying signals is covered in [Lesson 11](https://greatscottgadgets.com/sdr/11/), which will explain how to do this correctly.

closed 2022-04-21 by straithe