greatscottgadgets/hackrf #1073: hackrf_sweep reading binary output files

glaier, 2022-03-21:
I have difficulty finding documentation on the binary file format generated by hackrf_sweep.
ReadBin or similar functionality in R generates numbers that are not straightforward to interpret and tabulate with standard settings.
Basically I need to be able to reproduce the table in the current documentation of hackrf_sweep.

Best Wishes

glaier, 2022-03-22:
Here is a line of input from a binary file, read in R. 
The 420 is definitely the time stamp (4:20 am) I cannot make sense of the other integers.

> zz <- file('/home/gunnarhellmundlaier/5_6_GH.bin', "rb")
> vec<-as.numeric(as.vector(readBin(zz, "integer",53,size=8)))
> vec
 [1]         420           1           1 -1031137580 -1033999537 -1031656551
 [7] -1033606780 -1033129802 -1032356053 -1033893275 -1032363014 -1030810201
[13] -1034185255 -1033396624 -1034215557 -1033130023 -1031747402 -1030195928
[19] -1034231360 -1031714872 -1031779086 -1032505137 -1033796188 -1031994096
[25] -1033789994 -1033644195 -1033930948 -1033220004 -1034181647 -1032985638
[31] -1032485176 -1032144596 -1031601069 -1031674822 -1032418773 -1032666559
[37] -1032591480 -1031690839 -1033093319 -1030994021 -1033325393 -1031544436
[43] -1032892322 -1032291418 -1030941654 -1034189661 -1032660542 -1033326417
[49] -1031851353 -1031517110 -1031959594 -1033974117 -1033976206

martinling, 2022-03-22:
Hmm, looks like we're missing documentation for the binary format. The code that writes it is [here](https://github.com/greatscottgadgets/hackrf/blob/master/host/hackrf-tools/src/hackrf_sweep.c#L281-L297), so let me try and make sense of that.

If it were written out as a C structure it would look like this:
```c
struct record {
    uint32_t record_length; // 4 bytes
    uint64_t band_lower_edge_hz; // 8 bytes
    uint64_t band_upper_edge_hz; // 8 bytes
    float fft_data[length]; // 4 bytes each
}
```
The length in `record_length` is the length in bytes of the fields that follow it, so the `length` of the float array should be `(record_length - 16) / 4`.

I'm afraid I don't speak R, so I'm not sure off hand what the best way to extract those fields will be for you.

glaier, 2022-03-22:
I get the correct answers when I read in 4-byte sizes. Thank you. 

as.numeric(as.vector(readBin(zz,"numeric",53,size=4)))[6:53]
 [1] -52.59155 -69.04459 -58.49931 -55.60479 -56.36953 -65.08515 -63.26619 -57.10304
 [9] -56.84053 -58.92257 -61.55173 -61.87419 -64.34246 -56.01015 -55.56366 -61.84763
[17] -76.57574 -71.54229 -57.20924 -54.89634 -54.85207 -57.90472 -57.03611 -54.78074
[25] -54.95466 -58.92173 -63.56946 -64.39201 -69.98486 -76.22882 -65.17786 -54.72046
[33] -58.14720 -64.64020 -66.00278 -64.15028 -60.54185 -61.30548 -59.09333 -56.38051
[41] -59.95427 -63.25494 -56.35626 -56.40414 -63.54324 -56.96032 -56.77315 -55.86644

closed 2022-04-03 by glaier