greatscottgadgets/hackrf #1075: A possible data race without locking

ryancaicse, 2022-03-24:
Hi, it seems the `device->all_finished = false `at 257 is not protected by a lock?

https://github.com/greatscottgadgets/hackrf/blob/d3e4e9b6de2c94eb987fb4595b206eba8068c047/host/libhackrf/src/hackrf.c#L247-L257

martinling, 2022-03-24:
That lock is only needed to avoid a specific race between the transfer thread setting that flag, and the current thread going to sleep waiting for it to do so.

The context here is that we're cancelling transfers, so we're inside either `hackrf_stop_rx`, `hackrf_stop_tx`, or `hackrf_close`. Once we've cancelled the transfers, we wait for `all_finished` to be set by the transfer thread. We need a condition variable to wait safely, and that requires the use of the lock to protect the assertion being waited on. Once that condition has been signaled though, we know that there are no further transfers active and the transfer thread is idle, so it will not be changing any of these flags any more - that's why it's safe to reset them here.

The code that clears these flags is actually due to be removed by [this commit](https://github.com/greatscottgadgets/hackrf/pull/1071/commits/aef33c02396914b8d96bf926ef22491da04f1ed3) on PR #1071, which simplifies the completion tracking by replacing the flags with a single `active_transfers` counter.

Review on that version would be welcome, if this is an area of the code you're interested in.

ryancaicse, 2022-03-24:
@martinling Thank you very much for your detailed reply. Now, I clearly know this is safe. Feel free to close this issue.

martinling, 2022-03-24:
No problem. Your attention to detail is appreciated!

closed 2022-03-24 by martinling