greatscottgadgets/hackrf #1076: hackrf_sweep not transfering data after flashing older firmware

metayan, 2022-03-26:
### Steps to reproduce
1. Check out old version of hackrf (tried with 43e6f99fe8543094d18ff3a6550ed2066c398862 and 451873d616ba9bc6f73648d09445d812737a7c30)
1. Build firmware and host software.
2. Flash the old firmware `./hackrf-tools/src/hackrf_spiflash -w ../../firmware/build/hackrf_usb/hackrf_usb.bin`
3. Reset HackRFOne
4. Run `./hackrf-tools/src/hackrf_sweep -f 770:790 -w 250000 -N 1`

### Expected behaviour
Receive sweep data.

### Actual behaviour
Got `Couldn't transfer any data for one second.` instead of sweep data.

When the same firmware version is run with DFU from RAM, sweep data is received.
```
dfu-util --device 1fc9:000c --alt 0 --download ../../firmware/build/hackrf_usb/hackrf_usb.dfu
./hackrf-tools/src/hackrf_sweep -f 770:790 -w 250000 -N 1
```

### Version information
**Operating system**:
macOS Darwin 10.15.7

**libusb**
https://github.com/libusb/libusb/commit/780044d4e4248da81046063bac9cdd4721de8af5
https://github.com/libusb/libusb/commit/19826aaf493961e0287ef2f2006a4941f84f9537

**hackrf_info output for the failing flashed version:**
```
hackrf_info version: git-43e6f99f*
libhackrf version: git-43e6f99f* (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2919845f
Board ID Number: 2 (HackRF One)
Firmware Version: git-43e6f99f (API:1.04)
Part ID Number: 0xa000cb3c 0x00644f56
```

**hackrf_info output for the working DFU version:**
```
hackrf_info version: git-43e6f99f*
libhackrf version: git-43e6f99f* (0.5)
Found HackRF
Index: 0
Serial number: RunningFromRAM
Board ID Number: 2 (HackRF One)
Firmware Version: git-43e6f99f (API:1.04)
Part ID Number: 0xa000cb3c 0x00644f56
```

### Output
`LIBUSB_DEBUG=4 ./hackrf-tools/src/hackrf_sweep -f 770:790 -w 250000 -N 1`
```
Stop with Ctrl-C
[ 0.012890] [011f8818] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.012898] [011f8818] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.012900] [011f8818] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 500ms
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 0.05203 seconds (0.00 sweeps/second)
```

straithe, 2022-04-29:
Is there a specific reason you need an older firmware version, or does a newer version suffice?

Sheener22, 2022-04-30:
New firmware will be more than suitable. Thank you

metayan, 2022-04-30:
> Is there a specific reason you need an older firmware version, or does a newer version suffice?

I needed the old versions to verify that some fixes weren't introducing new problems.

Noticed the strange behavior and thought it might be indicating something not getting properly initialised.

straithe, 2022-04-30:
From what I understand, you are finding that you get ```Couldn't transfer any data for one second.``` with the old code. Is that correct?

metayan, 2022-04-30:
Yes, but only when it was flashed.
When I ran it from RAM, it worked fine.

The difference in behavior between flashed and run from RAM is what I found to be strange, so filed an issue, because I thought that maybe someone more familiar with the code might go "Aha, of course" and know right away what was causing it. ;)

straithe, 2022-05-01:
Do you have this issue with the current code?

metayan, 2022-05-01:
Not easy to say at this very moment, because I'd have to wait for a newer version to become available, and then flash the current version – as of this writing – to check.
A tricky one... ;)

straithe, 2022-07-14:
I'm confused by your statement. The text in your original issue post said that you checked out an old version of the HackRF software, and I'm asking you if the same issue happens when you use the latest software instead of an old version. Do you notice the issue you were experiencing with the newer software?

metayan, 2022-07-14:
I have not done the experiment with the latest software.
I needed to test some changes and happened to find that the two versions mentioned were behaving in one way when they were run from RAM and differently when they were flashed.

straithe, 2022-08-15:
I'm closing this issue as we have not been able to recreate this with the current code in the master branch. If it does occur again and you are on the master branch, please let us know!

closed 2022-08-15 by straithe