greatscottgadgets/hackrf #1077: Request sudo hackrf_reset

glaier, 2022-03-26:
"Reset the HackRF device by pressing the RESET button or by unplugging it and plugging it back in." or make a shortcut function running a software USB power off, wait, power-on cycle, i.e. https://stackoverflow.com/questions/4702216/controlling-a-usb-power-supply-on-off-with-linux.

There might be issues implementing on Windows demanding Administrator rights. On Linux, we should be able to call a short cut using sudo, because of the need introduced with hackrf_sweep and opera_cake.

# disable external wake-up; do this only once
echo disabled > /sys/bus/usb/devices/usb1/power/wakeup 

echo on > /sys/bus/usb/devices/usb1/power/level       # turn on
echo suspend > /sys/bus/usb/devices/usb1/power/level  # turn off

martinling, 2022-03-26:
Hi @glaier, could you clarify why it is that you need this?

If you just need to reset a connected HackRF you can do this with `hackrf_spiflash -R`. This sends a USB vendor request to the HackRF which tells the MCU to reset itself.

The firmware does have to know how to handle that request, but we've had that support for a long time now. The feature was added in #313 and first appeared in the 2017.02.1 release, so five years now.

When updating a HackRF that's running firmware older than that, it will be necessary to power cycle it or use the reset button.

glaier, 2022-04-01:
Thank you. It answers my question with respect to software.

A reset has been necessary when I use HamItUP. Maybe it is a question of using insulated USB cables and baluns dependent on antenna and frequency range. The build-up of static. Perhaps I will need a relay to turn my setup on and off working between frequency ranges and different setups.

closed 2022-04-01 by glaier