greatscottgadgets/hackrf #1078: Request hackrf_sweep bin size options

glaier, 2022-03-26:
hackrf_sweep -f 150:151 -w 40000 -l 16 -g 24 > /home/username/test_lowgain.csv
generates a file in which I cannot interpret results. 
The third and fourth column assumes bin size of 5MHz although I set frequency resolution to 40kHz and sweep range to 1MHz.
Please provide smaller bin sizes and/or sweep ranges, it makes sense in a world with opera_cake.

martinling, 2022-03-26:
Hi @glaier,

You should be getting 40kHz bins, as requested. The bin width is shown in column 5. In the first line of that file, the 25 values in columns 7 to 32 will be the 40kHz bin values from 150MHz to 151MHz. The remaining columns on that line contain further bin values, up to 155MHz.
```
# date      # time           # start    # end      # width        # bin values start here
2022-03-26, 07:12:11.502582, 150000000, 155000000, 40000.00, 500, -43.60, -56.97, -54.68, ...
```
You will then get three more lines, for 155-160MHz, 160-165MHz and 165-170MHz. This happens because the minimum tuning step for hackrf_sweep is 20MHz. You can just ignore the bins you're not interested in. The next data for 150-155MHz will appear on line 5.

glaier, 2022-04-01:
If I change the minimum tuning step in the source code? I want measurements in 0.1MHz bins... any recommendations? Do other settings need to be changed simultaneously?

I have changed to csv output since. 

The issue was the bin file format, not the interpretation of the output. 
I would recommend putting a python script example in the documentation for reading the binary output files into data tables. Most people would choose binary if it is easy to read in standard analysis software.

martinling, 2022-04-01:
> If I change the minimum tuning step in the source code? I want measurements in 0.1MHz bins... any recommendations? Do other settings need to be changed simultaneously?

If you want measurements in 0.1MHz bins, you just need to pass `-w 100000` to `hackrf_sweep` and you will get measurements in 0.1MHz bins.

The tuning step has nothing to do with the bin width. HackRF doesn't retune for each bin that you want to measure. It doesn't need to, because it can capture up to 20MHz of bandwidth at once.

Sweep mode always works in steps of 20MHz. When you ask for a sweep range of 150-151MHz, it only takes one step to cover that range. You'll get data for 150-170MHz, because that's what it collects in that one step. How many bins that range is broken down into depends on the bin width you set. At 0.1MHz, you'll get 200 bins. You can ignore the ones you're not interested in.

glaier, 2022-04-01:
Thank you Martin! So the 100khz bandwidth defines disjoint intervals or sub-bins, I was not sure of the interpretation. 

I have some further questions and remarks: Due to the harmonics of the electronics, I would believe powers of 2 to be more optimal bandwidth sizes? 

The question of optimal bandwidth should be seen in connection with the precision of measurements. Measurements at the beginning of the 20MHz interval have measurement precision of different sizes than at the middle or end of the 20MHz interval and also depend on the band. The precision depends on strength of the signal, distance from oscillator frequency on the board, etc. I believe it is of great importance when I compare several swipes in order to detect a signal. I would like to be able to do statistical tests and plot confidence and prediction intervals.

martinling, 2022-04-04:
> Due to the harmonics of the electronics, I would believe powers of 2 to be more optimal bandwidth sizes?

I'm not sure what you mean here. When receiving you will sometimes see spurs due to harmonics of internal signals, and you can sometimes move or avoid them by adjusting the center frequency and/or sample rate. I don't follow the connection you're drawing to powers of two, though.

> The question of optimal bandwidth should be seen in connection with the precision of measurements. Measurements at the beginning of the 20MHz interval have measurement precision of different sizes [...]

I don't quite follow what it is you're trying to do here, but given that you're looking for precise control of measurements at narrow frequency ranges, I suspect that `hackrf_sweep` is not really the tool you want.

The point of `hackrf_sweep` is to be able to scan a large amount of spectrum very quickly; its selling point is that it can sweep 6GHz in less than a second. It does that by having the device do most of the work in firmware. The device gathers ~16K of samples at each tuning step, sends all those sample blocks over USB, and then the `hackrf_sweep` program runs an FFT over each block to give you the output.

If you're looking at specific signals within a narrow frequency range, then there's no reason to use `hackrf_sweep`. You're going to be better off using HackRF in its normal receive mode. You can use `hackrf_transfer -r` to collect raw sample data and run the FFT, or whatever analysis you want, in your own code. Or you can use GNU Radio to build a flowgraph that does what you need. Doing things this way will give you full control of all the parameters involved.

glaier, 2022-04-20:
With respect to instrument precision. It seems to be common for electronics freaks and amateur radio to measure precision at a range of frequency with a given input from an oscillator or a set of oscillators. Such measurements will provide the knowledge I am seeking concerning "variability". I realize in reality there is also a dependence on the antenna needing separate testing in the laboratory, however I believe the precision of the HackRF should be made available in the manual along the frequencies from 1MHz to 6GHz.

straithe, 2022-04-28:
This really isn't something we can provide.

Great Scott Gadgets manufactures and sells HackRF One as uncalibrated test equipment; the hardware for each individual unit is tested to verify that its performance is within an acceptable range. HackRF intended to be useful for experimentation, testing and learning, not for making precise measurements. If you are looking for precise measurements I suggest you look at Keysight or R&S as those devices are calibrated hardware explicitly intended for extremely precise and accurate measurements. 

You could develop your own calibration process for your individual HackRF device's performance, by measuring it against calibrated reference equipment. With some further statistical work, you could estimate the precision of your measurements. This would take a lot of work, but the results would be specific to that individual device, with those settings, under those conditions - and would probably not remain correct, because calibrated test equipment is specifically designed to ensure long term stability and repeatability, in ways that HackRF is not.

The closest HackRF can get to what I think you are trying to do is X. This is not something that can be fixed in software.

closed 2022-07-14 by straithe