greatscottgadgets/hackrf #1084: Portapack no longer receiving radio signal in HackRF mode (following a minor gravitational incident)

endjuser, 2022-04-03:
Hello,

I have a portapack H2 recently updated with Mayhem 1.4.4  This morning the device accidentally fell about 6 inches while connected to my laptop and thereafter stopped receiving any radio signals over USB.

**Any ideas on what I can check for failure?**

- This had been a fully-working installation before the device fell

- Everything still works normally in portapack standalone mode -- I can tune into FM radio, capture ADS-B, etc. and see expected changes in the spectrum analyzer waterfall as I move the antenna around -- so the internal radio hardware seems to be working OK

- When I put the device into HackRF mode, it is recognized by my PC.  I am able to connect to it with SDRSharp and adjust gain/etc.  I see the frequency display change on the portapack as expected as I adjust in SDRSharp -- so USB data connection between PC/device seems to be working OK

- When in HackRF mode, the only signal visible on SDRSharp waterfall is the DC voltage spike in the center and two reflections, regardless of frequency or gain settings. Using hackrf_transfer with a strong FM radio frequency outputs mostly FE, FF, 00, or 01 with an occasional 02.

Things I have tried already:
- tried other USB cables (no change)
- tried my other non-portapack HackRF (works OK -- so the laptop and USB cable are not an issue)
- tried on my linux laptop with gnuradio (no change)
- tried re-flashing the portapack firmware over USB (completed successfully, but no change)

### Steps to reproduce
1. Connect portapack to PC in HackRF mode
2. Drop your portapack off of a box onto your table (approx. six inches)

### Expected behaviour
None, except perhaps uttering some choice words

### Actual behaviour
Radio signals no longer received in HackRF mode

### Version information
**hackrf_info output:**
hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000614065c8356e7543
Board ID Number: 2 (HackRF One)
Firmware Version: local-57ed3eb (API:1.04)
Part ID Number: 0xa000cb3c 0x00614354

martinling, 2022-04-03:
I'd suggest starting by eliminating a bunch of variables - remove the HackRF board from the Portapack, flash the normal HackRF firmware onto it, and see how things look then.

endjuser, 2022-04-04:
Updated original post after some further testing (and removing the HackRF board/flashing normal firmware).

Apparently it works in HackRF mode if I turn the AMP OFF, but with the AMP ON it receives nothing in HackRF mode.  In portapack mode turning the AMP ON appears to reduce the signal strength, although I am still able to receive FM radio, ADS-B, etc. whether it is ON or OFF.

endjuser, 2022-04-07:
Thanks @wolfpcgn
I don't entirely understand how that would have happened (I was decoding RDS from FM radio using the standard antenna at the time of failure), but it does seem like the LNA is malfunctioning.  I ordered some spares the other day and will update here once I've tried replacing them

endjuser, 2022-04-14:
I replaced the LNA with a new MGA-81563 part and that resolved my issue.

closed 2022-04-14 by endjuser

straithe, 2022-04-14:
Thank you for the update