greatscottgadgets/hackrf #1086: Maximum return power during TX ?

Batchyx, 2022-04-05:
I'm considering hooking an HackRF to transmit to another RF device using an RF cable and some attenuators¹.

Unfortunately the HackRF and the device may transmit at the same time.

What amount of input power can the HackRF tolerate while transmitting ?
I need this to select how much attenuation I need, among other things.

⁽¹⁾ The actual setup is probably going to be more complicated.

martinling, 2022-04-05:
I'm not sure we have any numbers for what a safe return level might be during TX.

As a lower bound, you would be fine at -5dBm, which is the RX input power rating.

If you need to go higher than that, you're going to need to be extremely careful about what configurations the HackRF signal path could be in when the other device transmits, because that will determine which components are exposed:

[image]

Which components are exposed will depend on whether the amplifiers are in use or bypassed, whether the mixer is in use or bypassed (which depends on your operating frequency: 2175 to 2750MHz bypasses the mixer), and also whether the HackRF is guaranteed to be in TX mode at all times when the other device may transmit.

Batchyx, 2022-04-05:
So -5dBm should be ok, even if the HackRF transmits at a lower power ?

The frequency is around 5 GHz so the mixer is in use, and from the info I gleaned, The HackRF should only be able to transmit up to around -10 dBm at this frequency, and I will probably have to transmit close to the max because of the attenuators (the other device transmit at a higher power... ), so the tx amp is in use.

The HackRF is expected to transmit often enough (every 1ms or so) that I don't think I can toggle it off instead of transmitting zeroes.

Also i tried looking at the datasheet for the various components (e.g. MGA81563 or MAX2837) but I cannot find anything looking like max return power in them. Probably I do not understand them.

martinling, 2022-04-05:
OK, if you're staying in TX mode throughout with the amp enabled, then the question is just about safe power levels at the output of the MGA81563.

I don't think we have any information about that other than what's in the datasheet, and the relevant figure there is the absolute maximum output voltage relative to ground, which is 6V.

We run the amp on a 3.3V supply and that's also the DC bias of the output. So to stay within the maximum rating, the output can't swing upwards by more than 2.7V, which corresponds to 5.4V peak-to-peak.

At 5GHz, HackRF's output power will be somewher in the -10dBm to 0dBm range - let's assume to be safe that you have a sporty one that's at the higher end of that range. A power level of 0dBm corresponds to a peak-to-peak swing of 0.63V in a 50 ohm system.

So if we assume for simplicity that the output and return signals superimpose as added voltages, the return signal can have a swing of 5.4 - 0.63 = 4.77V before it would exceed the absolute maximum voltage at the amplifier output. That swing corresponds to a power of 17.4dBm.

I would immediately knock several dBs off that figure, both because you want to stay well away from the absolute maximum rating, and because the reality of how the two interact may not be anywhere near that simple.

But it seems likely that up to 10dBm or so would be OK.

Batchyx, 2022-04-05:
Thanks for the great and quick answer !

Should I close this issue ? or should it be documented somewhere ?

martinling, 2022-04-05:
Given that my answer is extremely specific to your particular scenario, I don't think this really belongs in the documentation, so I'll go ahead and close this.

closed 2022-04-05 by martinling