greatscottgadgets/hackrf #1087: Burnt component - Finding replacement

helmet21, 2022-04-13:
### Steps to reproduce
1. Plug in USB cable
2. LEDs first behave normally and after a few seconds they dim down and start flickering
3. A component on the board gets very hot (instantly goes to 47 degrees celsius)

### Expected behaviour
Normal operation

### Actual behaviour
lsusb shows no device

### Version information
**Operating system**:
Both on Windows 10 and Ubuntu 20.04.

**hackrf_info output:**
Says no device found.

---

I'm trying to identify the component since I assume that it is burnt probably due to me running voltage through the RF input port.

[image]

This is the component that gets hot. The writing on it says "CAT OCI AVHR" but I can't find any information about it online and I can't figure out which one it is on the schematic. Can anyone tell me how to find this component so that I can try to replace it?

Best regards
@helmet21

martinling, 2022-04-13:
That is part U21 on the schematic. It's a TPS62410DRCR regulator.

helmet21, 2022-04-19:
Thanks will order the part and give it a go.

closed 2022-04-19 by helmet21