greatscottgadgets/hackrf #1088: just a question about step

j2l, 2022-04-16:
Reading the code
```
#define DEFAULT_SAMPLE_RATE_HZ (20000000) /* 20MHz default sample rate 
--
#<span class="pl-k">define</span> <span class="pl-en">TUNE_<span class="text-bold hx_keyword-hl rounded-2 d-inline-block">STEP</span></span> (DEFAULT_SAMPLE_RATE_HZ / FREQ_ONE_MHZ)
```
I understand that for a 4Mhz band, the step is 20000000÷4= 5Mhz
Is that correct?

I'm asking because I'm looking for a 10Hz step like high-end radios to tune to weather fax on boat and still use this for VHF.

straithe, 2022-04-29:
Thank you for reaching out. Unfortunately, I don't understand what you are saying or what your question is. Can you please try rewording it for me?

j2l, 2022-04-29:
Thanks for your reply @straithe 
Local marine weather forecast is sent over radio (frequency depends on the location of the boat on earth).
For instance, in your country, you could tune to the HALIFAX station at 4271 kHz (4.271Mhz).
To tune to (and correctly isolate) 4271khz, the maximal step here must be 1 kHz max.

Audio code is very sensitive, if you don't have the correct tuning, you'll here something but you won't get the image right.

From the code extract above, it looks like the sample rate is varying according to the frequency, hence the step (smallest chunk of frequency distinguished from the closest chunk).
(si vous parlez aussi français, je parle ici du pas (step) et de l'échantillonage (sample rate)).
If for 4Mhz, the sample rate of the HackRF One is 5Mhz (the spectrum of listened frequencies, visible in your software), the step might be far above 1khz (even if you can enter 4'271'000 on your software like SDR#).

Could you (or anyone here using a HackRF One device) could tune in the local weather forecast and try to get a forecast image to post here?

Here are frequencies all over the world:
STATION | Région | Fréquence
-- | -- | --
CHARLEVILLE | AUSTRALIA | 2628 kHz, 5100 kHz, 11030 kHz, 13920 kHz, 20469  					 kHz
WILUNA | AUSTRALIA | 5755 kHz, 7535 kHz, 10555 kHz, 15615 kHz, 18060  					 kHz
RIO DE JANEIRO | BRAZIL | 12665 kHz, 16978 kHz
HALIFAX, NOVA SCOTIA | CANADA | 4271 kHz, 6496.4 kHz, 10536 kHz, 13510 kHz
VALPARAISO PLAYA ANCHA | CHILE | 4228.0 kHz, 8677.0 kHz, 17146.4 kHz
BEIJING (PEKING) | CHINA | 5526.9 kHz, 8121.9 kHz, 10116.9 kHz, 14366.9 kHz,  					 16025.9 kHz, 18236.9 kHz
TAIPEI | CHINA | 4616 kH, 8140 kHz, 13900 kHz, 18560 kHz
HAMBURG | GERMANY | 3855 kHz, 7880 kHz, 13882.5 kHz
ATHENS | GREECE | 4481 kHz, 8105 kHz
KYODO NEWS AGENCY | JAPAN | 4316 kHz, 8467.5 kHz, 12745.5 kHz, 16971 kHz,  					 17069.6 kHz, 22542 kHz
TOKYO | JAPAN | 3622.5 kHz, 7795 kHz, 13988.5 kHz
SEOUL | KOREA | 3585 kHz, 5857.5 kHz, 7433.5 kHz, 9165 kHz,  					 13570 kHz
MURMANSK | RUSSIA | 5336 kHz, 7908.8 kHz, 10130 kHz
BANGKOK | THAILAND | 7395 kHz
NORTHWOOD | UK | 2618.5 kHz, 4610 kHz, 8040 kH, 11086.5 kHz
KODIAK, ALASKA | USA | 2054 kHz, 4298 kHz, 8459 kHz, 12412.5 kHz
PT. REYES, CALIFORNIA | USA | 4346 kHz, 8682 kHz, 12786 kHz, 17151.2 kHz,  					 22527 kHz
HONOLULU, HAWAII | USA | 9982.5 kHz, 11090 kHz, 16135 kHz
NEW ORLEANS, LOUISIANA | USA | 4317.9 kHz, 8503.9 kHz, 12789.9 kHz, 17146.4 kHz
BOSTON, MASSACHUSETTS | USA | 4235 kHz, 6340.5 kHz, 9110 kHz, 12750 kHz

martinling, 2022-04-29:
Hi @j2l,

The `TUNE_STEP` you're referring to is part of the `hackrf_sweep` code. It's not a tuning resolution of the radio, but rather a step size used in sweep mode, i.e. when HackRF is working as a spectrum analyzer, rather than a receiver.

In that mode, we capture 20MHz of bandwidth for a brief period and then tune upwards by 5MHz and repeat. It's 5MHz rather than 10MHz because we discard the middle and the edges of each 20MHz block to avoid the filter edges and the DC spike in the middle. The point of this mode is to quickly gather a power spectrum across a wide bandwidth - we sweep 0-6GHz in 0.75s.

When receiving you can set any tuning frequency you want. Internally there are multiple oscillators that the firmware will configure to get as close as it can to what you request. I'm not sure off-hand what the effective step size is (it will vary somewhat with frequency) but it's not something that should ever really be a limitation in practice.

With an SDR receiver it's normal to just tune somewhere near the frequency you want - you're collecting up to 20MHz of bandwidth around that point. Picking out individual signals is normally done in software. In fact you generally don't want to tune exactly to the frequency of your signal of interest because that will put a DC spike there - an artifact inherent to quadrature sampling systems.

j2l, 2022-04-29:
Thank you @martinling for this clear explanation! 
Indeed, I was confusing the spectrum analyzer and the software part.
I'd need to validate weather fax reception is good enough before buying the HackRF One (also to use it for other frequencies).
Do you think it would be possible to test it on your side?

straithe, 2022-06-24:
We do have a lot underway at the moment, so it is not likely that our team will have time to test this on our side. Please let us know how your testing goes.

closed 2022-06-24 by straithe