greatscottgadgets/hackrf #1092: pyhackrf lib

Ayoubdz19, 2022-04-20:
### Steps to reproduce
1.i'm looking for a library wish can allow me to control HackRF sdr with python script to get the signal at the output the do some signal processing with python. 
2. i downloded this project https://github.com/dressel/pyhackrf but it didn't work for me. 
### Expected behaviour
get the signal at the output of the sdr 

### Actual behaviour
i could'nt install the pyhackrf lib 

### Version information
Windows 
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2022-04-28:
I recommend opening an issue on https://github.com/dressel/pyhackrf since that is a repository we at GSG are not involved in.

straithe, 2022-04-28:
You may also want to check out: https://wiki.gnuradio.org/index.php/Guided_Tutorial_GNU_Radio_in_Python instead.

closed 2022-07-14 by straithe