greatscottgadgets/hackrf #1096: Is there a software way in GNU Radio to disable MAX2837 and the RF front-end for ADC-only operation?

goscickiw, 2022-05-01:
I would like to be able to use the HackRF as a 2-channel baseband ADC/DAC directly through the P9 connector. Is there a software way to stop the MAX2837 transceiver and the rest of the RF front-end from turning on when receiving or transmitting starts, so it doesn't interfere? I have noticed on the schematics that there are lines going from the main processor to the MAX2837 for enabling RX, TX, and the whole transceiver, so the processor has control over that.

Is it possible using GNU radio osmocom source and sink? There is the "Device arguments" field, but from the osmocom documentation it seems that it can only be used to set the device ID, enable the bias voltage, or change the buffer setting. There is also a gray "command" input on the osmocom GNU Radio block. I haven't used these gray inputs before and I don't know how they work yet.

Also, would it then be able to operate full duplex with osmocom sink and source? Are there any specific things I need to do to make it work other than just setting up the sink and source as normal?

martinling, 2022-06-01:
There's no support for doing this in the standard firmware. You could patch the firmware so that it never enables the MAX2837, by removing the `gpio_write` calls from `max2837_target_set_mode` in `firmware/common/max2837_target.c`.

However, that would still leave the MAX2837 `RXBB` outputs connected to the MAX5864 and the P9 connector. I can't see from the datasheet if it's safe to drive those with an external signal while the MAX2837 is in shutdown mode. I think it must be okay though because that's what the Neapolitan full-duplex add-on board does.

goscickiw, 2022-06-01:
Thanks, I will try that, but still it would mean that I'd have to switch between different firmware versions depending on if I want to use the RF front-end or not.
Would it be possible to add a way to toggle this in software in the future?

closed 2022-07-14 by straithe