greatscottgadgets/hackrf #1099: External clock not recongnized with

gauxonz, 2022-05-14:
### Steps to reproduce
1. Power on my hackrf one without a TCXO then run `hackrf_debug --si5351c -u 1 -n 0 -r`, I got '[  0] -> 0x50'
2. Plug a 10Mhz TCXO module and reboot the hackrf, run `hackrf_debug --si5351c -u 1 -n 0 -r` again, the outcome stays the same  '[  0] -> 0x50'
[image]

### Expected behaviour
Accroding to the [wiki](https://hackrf.readthedocs.io/en/latest/external_clock_interface.html), the outcome of `hackrf_debug --si5351c -u 1 -n 0 -r` should be either `0x51` or `0x01` (Is this correct ?)

### Actual behaviour
The result is always `0x51`

### Version information
**Operating system**:
Linux gz-PC-01 5.15.0-30-generic #31-Ubuntu SMP Thu May 5 10:00:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000723864c02e58a08f
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00614761

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

simonboydfoley1, 2022-05-15:
Same issue it may be firmware related as it's supposed to read 0x51 not 0x50 if the TCXO is not detected. Perhaps things have changed With The latest firmware release. I checked the power and the External clock was getting 3.1v. So I will get a scope on the clock and make sure it's working as expected. Turns out mine was a clone sold by Amazon on eBay that was second hand and had had some reflow work done on it and the Design Fault on the Amp chip had occurred and it was blown too so I'm sending it back. I'm probably going to get a Pluto Plus because that design fault will just keep on popping the Amp as I have an internal VHF amp in my house to gain the signal to my shack room / office. I'll let you know if I get it working.

miek, 2022-05-15:
I saw a similar issue with another user recently, some clone HackRFs seem to be shipping without CLKIN support.

An extra unpopulated footprint has been added that breaks the CLKIN track. I think I can just see the edge of it on your board, but you can double check if it looks like this:

[image]

The other user tried populating that footprint but it didn't solve the issue.

This is just a theory at the moment, but CLKIN may have been disconnected intentionally if those boards have a Si5351A/Si5351B populated (which do not have an external clock input) instead of the Si5351C in the original HackRF design.

----

> it may be firmware related as it's supposed to read 0x51 not 0x50 if the TCXO is not detected. Perhaps things have changed With The latest firmware release.

It's not related to firmware. The reason it's returning `0x50` instead is because the populated clock IC is returning a different `REVID` field, which suggests that they may be old stock of Revision A parts.

gauxonz, 2022-05-15:
@miek You are correct. I checked my chip, it's a si5351-A, and the CLKIN track is broken too. 
[image]

I've checked this issue with the vendor, and they admitted this defect. 
My vendor says:
> the pandemic raises the price of Si5351C, so some of this chip is replaced by rev. A. 
, which is aggravating.
So the conclusion is that one should check the chip model before making the receipt.

closed 2022-05-15 by gauxonz

simonboydfoley, 2022-05-15:
Yup not only did I have a blown Amp Chip I also have the same Chip Si5351  Missing the C at the end so the bundle I bough off Amazon including the TCXO could not have possibly have worked ......

gauxonz, 2022-05-15:
@simonboydfoley Crap I didn't check my Amp chip... What actually is the design fault you mentioned?  How do you find it?

simonboydfoley1, 2022-05-15:
The card is susceptible to having the Chip that delivers the power amp from
being over volted and fried by the input signal. You can tell if you have
blown the MGA-81563 <https://amzn.to/2XpvNj5> chip.
See below ....but basically when you enable the amp feature in software
like SdrSharp it will actually result in the strength of the signal
dropping, not raising and you will loose the station you are tuned into ...

https://www.onesdr.com/2020/04/09/how-to-fix-a-broken-hackrf/

simonboydfoley, 2022-05-17:
I have ordered a genuine retail HackRF One from MLS. I regret ordering form Amazon, the clone was horrific, when I opened the case I found flux all over the board around the PCB 2.54mm Female Pin headers so it was not wave soldered but done manually. There was solder blobs on the PCB that had to be picked and cleaned off. The MGA-81563 had been blown so the Amp was dead from the get go.  just not work the saving IMHO .... you go clone you accept the risk you get usless crap with poor QA. #lesson learned

hamidghafi, 2022-09-21:
I have the same problem as @gauxonz. 
I need to have a shared clock for four number of hackrfs. Unfortunately, the boards have si5351A chips like mentioned above.  I have an old hackrf with si5351B in which clock-in works. Is it possible to buy a si5351B (or C) chip and replace it with the current si5351A? Another solution which I would like to know your idea is that, I read somewhere that the si5351A accept clock-in of frequencies higher than 25 MHz. Is it possible to feed all clock-ins with a 25 MHz external clock(splitted)?

straithe, 2022-09-21:
@hamidghafi can you please open a new issue with your question? It'll be easier for our technical support team to keep track and get back to you.

simonboydfoley, 2022-09-21:
@hamidghafi you would need to pull the schematics of the chips and check to see if there is any additional logic circuits needed to feed into the  Si5351C to enable it to use an external clock (e.g. Pull Up/PullDown resistors). If your board was built around a  Si5351 or potentially a  Si5351B there may not be all the traces you need to simply replace the  Si5351A/B with the  Si5351C chip. 
But if the only difference between the chips is the removal of support for an external clock you may be able to do a chip replacement. Even if you were missing a trace for the external clock you could run that in externally with enammel wire.
I did thik of doin that and I did track down a chip on RS Components ..... but then I realised I coudl just return the buggerd clone board and buy an original.