greatscottgadgets/hackrf #1100: Do not confuse the power injector connectors when connecting

H6LS1S, 2022-05-30:
### Steps to reproduce
1. Connect the power injector ( for Mini Whip )  to the HackRF One with the reverse side
2. Apply 9V power
3. Congrats you broke something

### Actual behaviour
I collected all the necessary parts for the Mini Whip antenna and confused up the inputs on the power injector. HackRF began to whistle strangely (similar to the whistle of chokes or an electrolytic capacitor). Without the injector HackRF works longer and even outputs some kind of signal in CubicSDR. But after a while (10 seconds + - ) the USB turns off. Two coils near the USB input (marking *H) get very hot. 

By LED indication:
- constantly 3v3
- flickers 1v8 and RF
- the other are disabled

Can you please tell me what I broke?

### Version information

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc299656c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00664358

martinling, 2022-05-30:
The bad news is that all sorts of things could be damaged. Excess voltage at the RF port can get onto the 3.3V rail through the body diode of Q4 in the bias tee circuit. In theory the resulting voltages can fry everything on the board.

The good news is that in practice, it seems that often either the regulator (U21) or the USB TVS array (U15) will clamp the downstream voltages to a low enough level to prevent damage, burning themselves out but saving the rest of the components.

The coils you mention getting hot are associated with the regulator (U21) - it's the chip between those two coils, and is probably where the heat is coming from. In at least one case (#668) a board with the same symptoms could be salvaged by replacing U21. However, it's also possible that the regulator is only getting hot because components downstream of it have failed, and are now drawing too much current from the regulator.

Unfortunately, that part (TPS62410DRCR) is out of stock in most places right now, so it might be hard to replace.

H6LS1S, 2022-05-31:
I was able to find TPS62410DRCR it at my place, in Kharkiv UA, I hope the warehouse was not damaged - http://www.kosmodrom.com.ua/prodlist.php?name=TPS62410

martinling, 2022-05-31:
That's great news to find some there, I hope the supplier is OK and been able to get back to business. Let us know how you get on.

closed 2022-06-13 by straithe