greatscottgadgets/hackrf #1104: HackRF Sudden  Hardware Failure

chakid21, 2022-06-10:
### Steps that Lead to Issue
1. Turned on HackRF & Portapack H2+ combo (that's been in light use for 1 month).
2. Turned device on and went to the main menu, no problem.
3. Walked away, came back.
4. Hackrf dead, no lights, no screen output on portapack, portapack power lights still.
5. Portpack responds to turning on and off, hackrf does not. Separated the portapack and plugged in hackrf direct with usb; nothing, no LEDs illuminate and the computer doesn't recognize any USB devices (tested multiple wires and 2 computers with different OSs).

### Expected behavior
The ability to power on the HackRF

### Actual behavior
HackRF seems hard bricked.

### Version information
**Operating system**: Ubuntu 21.10 & windows 7

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.

### Third Party Software 

[Mayhem firmware v1.5.1 ](https://github.com/eried/portapack-mayhem/releases/tag/v1.5.1)

### Comment

I have no idea what caused this issue. I'd like to hope it's something as simple as a fuse but I need assistance troubleshooting. Adafruit (the merchant) refused to help troubleshoot and said they would only warranty it if someone from Great Scott Gadgets can say it's defective. Any help or support is greatly appreciated.

straithe, 2022-06-13:
Can you post a picture of your HackRF (without the Portapack on it) here?

chakid21, 2022-06-13:
[image]

straithe, 2022-07-14:
It look like this may be a hardware failure. Please contact your reseller for a replacement.

closed 2022-07-21 by straithe