greatscottgadgets/hackrf #1105: AMD CPU's Instruction set - HackRf does not work

Rickmarben, 2022-06-12:
### Steps to reproduce
1. When HackRf is connected to machines with Intel CPU's the USB comes on and it appears on device manager OS is MS Windows 10
2. When HackRF is connected to the same built as in point 1 but with a Ryzen 5 5600 HackRF does not lit up the USB and the OS is Windows 10. Motherboard is in both instances ( 1 and 2) ASUS Prime B450 Plus
3. We then took a Dell Laptop with i7 10th generation and windows 10 and it works as described in point 1
4. Could it be that the way it has been designed will not work with ANY AMD cpus ?????
5. HackR was also updated with latest firmware but only works on INTEL CPU's

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2022-06-13:
I'm a little confused. The ASUS Prime B450 Plus is a motherboard for AMD not Intel, so I don't understand how you are using the same motherboard in both instances. Can you please give more detail about your test set up?

Rickmarben, 2022-06-13:
Hi, thanks for coming back. Ok, I have two different physical machines. mine and my wife's. Both have been built with the same hardware ( I did it). Since I run Linux Mint 20.3 and my wife's runs WIN 10, Initially I thought could be OS related ( OS= Operating System). So after that I used 2 other machines from my 2 sons. They both are Intel CPU's and then we tried with an old dell laptop with i5 Intel and the last was my work Dell laptop with i7 10th Gen Intel CPU.
At that point, since my machine and mi wife's are both Ryzen 5 3600 AMD, I concluded that something smells rotten in Denmark and could be chip instruction set related, but I am not a programmer.....................cheers

straithe, 2022-06-13:
HackRF is designed to work with AMD, so that is very likely not the issue you are experiencing.

Do any of the other lights come on when you plug in your HackRF?

Rickmarben, 2022-06-13:
Thanks, ok, very easy:
1) With the AMD machines ( either WIN OS or Linux), on power up the USB blinks on and then goes off and never lit up again.
2) On the INTEL ones no dramas.

Mind due the INTEL machines have different motherboards.
In the case of the AMD cpu's both machines are ASUS Prime B450 Plus.

martinling, 2022-06-14:
The CPU differences are definitely not related.

For the USB LED to light up on the HackRF doesn't require any of our software to run on the machine. It just requires the OS to take the USB device out of reset and put it into its single, default configuration. Any OS that sees the device on the USB bus should do this.

It's worth checking for power issues as @metayan mentioned, but it seems very unlikely that the HackRF wouldn't be getting enough power from any modern motherboard.

My best guess is that we're dealing with some intermittent fault or startup problem on that HackRF, and that by chance, it works OK with some machines - perhaps down slight differences in voltage or ripple on the USB 5V supply.

It's possible that this is actually an instance of the PLL startup issue that we just fixed in #1106, so it might be worth updating to the latest firmware from git master.

Rickmarben, 2022-06-14:
Metayan, Martinling, thanks. 

1) Anything I have at home works on USB except the HackRF/
2) But............my son has found a very interesting article (at work now do not have it handy) that the ASUS B450 motherboard and AMD cpu's series 3000 and 5000; the combination of both has problems with users saying intermittent USB dramas.

I will get the link later and put it here. Something somewhere is funny and would love to nail it down.

martinling, 2022-06-14:
That's interesting, but the intermittent dropouts described don't sound like the same issue as this single device repeatedly failing to enumerate.

I have one machine with an Asus B450 board, and haven't had any problems using a HackRF with it.

Rickmarben, 2022-06-15:
Martinling, do you have an AMD CPU in that motherboard ? What version of firmware are you running ? 
Tend to agree, something does not click in place. As soon as you physically plug it, the light momentarily comes on and then is off forever.
I may have to update firmware and try again. A total mystery.............................. Since I built both machines whatever it had from out of the box, is still there. I never like to change firmware for the sake of the latest ones. Never touch things that are working properly.

straithe, 2022-06-15:
What @martinling was trying to suggest is that you update the firmware on HackRF, not the firmware for your computer.

Rickmarben, 2022-06-15:
Ooopssssss, yes, that makes sense. That is easy to do.

Rickmarben, 2022-06-17:
Good morning guys, sorry been busy.

1) Martling says he has a B450. There are many versions of it, not only one. Secondly it has to be an AMD 3000 or 5000 series where that combination of problems start. Very clear in the ASUS site that there is no one single version of B450 and I have said which one I have. Just in case I need to repeat, USB light momentarily goes on then never comes back. So that is an intermittent drama as described in the link I have posted before.

2) Yes, we have put the very latest available firmware in the HackRF and the problem persist. So something else is going on. I do not know what it is.

Rickmarben, 2022-06-17:
Just in case, there are at least 9 versions of the B450
https://www.asus.com/microsite/motherboard/Best-AMD-Ryzen-B450/

Rickmarben, 2022-06-17:
PROBLEM SOLVED. We are now Transmitting DVB-T2 with GNU Radio to our TV.
I am and have been for many year a radio amateur operator..............................last year the radio would not work on USB to the PC................changed USB cable........
Guess what????? The USB cable that came with HackRF..............................have another 3 cables from mobile phones...................good bye drama. UNREAL !!!!!!!!

straithe, 2022-06-17:
Thank you for the update.

closed 2022-06-17 by straithe