greatscottgadgets/hackrf #1107: How to build host software on Windows

stevewillson, 2022-06-15:
I'm writing these instructions to document a successful build process of the host software on Windows 11 using MSYS2. I had trouble following the build instructions on the [README](https://github.com/greatscottgadgets/hackrf/blob/master/host/README.md#how-to-build-host-software-on-windows)

# How to build host software on Windows:
## Prerequisites:
* MSYS2 (https://www.msys2.org/)

MSYS2 Packages:
* `base-devel`
* `mingw-w64-x86_64-toolchain`
* `mingw-w64-x86_64-cmake`
* `mingw-w64-x86_64-fftw`
* `git`

Open `MSYS2 MinGW x64` from the Start Menu

Update the package repository
`$ pacman -Syu`

Install the required packages
`$ pacman -Syu --needed base-devel mingw-w64-x86_64-toolchain mingw-w64-x86_64-cmake git`

Download the libusb pre-compiled binary files:
https://github.com/libusb/libusb/releases

I used version [1.0.26](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26-binaries.7z)

Extract and copy the `libusb-1.0.26-binaries` folder to:
`/usr/local/include`

Clone the hackrf repository

`$ git clone https://github.com/greatscottgadgets/hackrf.git`

`$ cd hackrf`

`$ mkdir host/build`

`$ cd host/build`

Create the Makefiles
`$ ${MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/usr/local/include/libusb-1.0.26-binaries/libusb-MinGW-x64/include/libusb-1.0/ -DLIBUSB_LIBRARIES=/usr/local/include/libusb-1.0.26-binaries/libusb-MinGW-x64/bin/msys-usb-1.0.dll -DFFTW_LIBRARIES=/lib/`

`$ make`

`$ make install` (may require running MSYS2 as Administrator)

This will install the hackrf host tools to:
`C:\Program Files (x86)\HackRF\bin`

When I tried to execute `hackrf_info.exe` from that directory, I had to add the following folders to the Windows `Path` Environment Variable to locate the DLLs.
Add the following entries to the Path:
* `C:\msys64\mingw64\bin`
* `C:\msys64\usr\local\include\libusb-1.0.26-binaries\libusb-MinGW-x64\bin`

Open a Windows CMD Shell and execute:
`C:\Program Files (x86)\HackRF\bin\hackrf_info.exe`

If a HackRF board is connected, the information should be displayed.

martinling, 2022-06-15:
Thanks for writing this up! I'm reopening this to remind us to update the documentation.

corcorj, 2022-07-01:
After I get past moving to the host/build directory step and try to create the make files with the shown code i get an output like this:

MSYS ~/hackrf/host/build
$ ${MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/lib -DFFTW_LIBRARIES=/lib
bash: /bin/cmake: No such file or directory

I assume it has something to do with the path and variables however I have: C:\Program Files\CMake\bin , C:\Program Files\mingw-w64\x86_64-7.3.0-posix-seh-rt_v5-rev0\mingw64\bin , AND C:\opencv\build\install\x64\mingw\bin

In my PATH environment could someone help me out with this?

corcorj, 2022-07-02:
thank you for that clarification that helped me overcome that specific line of code however now when I moved onto the make and make install commands I got errors.. 

It looks like I didn't have libusb.h in the right directory on the make and make install commands but when I generated the make file with the command I was missing the $ with, looking at the output it does show that I have libusb installed correctly for that command to work correctly and output a make file or build files to : C:/msys64/home/jcorc/hackrf/host/build.

 I installed libusb with Cygwin and I also followed the link in the documentation listing it as a prereq to be able to install hackRF Software, however the download link downloaded a linux file and i am on windows the name of that file was: libusbx-1.0.18 (1).tar.bz2 

Is the Cygwin install of libusb not sufficent for the needs of the hackRF tools installation? I googled more windows friendly libusb installers and ran them as well but these errors I am getting are leading me to believe that a file or directory wasn't created for which the make install command was looking for..

Any thoughts or ideas as to why I am getting the errors I got when running the make and make install commands outputed below??

Thank you

Here is my commandline output:

jcorc@Beast MINGW64 ~
# cd hackrf

jcorc@Beast MINGW64 ~/hackrf
# cd host/build

jcorc@Beast MINGW64 ~/hackrf/host/build
# /usr/bin/cmake -version
bash: /usr/bin/cmake: No such file or directory

jcorc@Beast MINGW64 ~/hackrf/host/build
# /mingw64/bin/cmake -version
cmake version 3.23.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

jcorc@Beast MINGW64 ~/hackrf/host/build
# /mingw64/bin/cmake -h
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Options
  -S <path-to-source>          = Explicitly specify a source directory.
  -B <path-to-build>           = Explicitly specify a build directory.
  -C <initial-cache>           = Pre-load a script to populate the cache.
  -D <var>[:<type>]=<value>    = Create or update a cmake cache entry.
  -U <globbing_expr>           = Remove matching entries from CMake cache.
  -G <generator-name>          = Specify a build system generator.
  -T <toolset-name>            = Specify toolset name if supported by
                                 generator.
  -A <platform-name>           = Specify platform name if supported by
                                 generator.
  --toolchain <file>           = Specify toolchain file
                                 [CMAKE_TOOLCHAIN_FILE].
  --install-prefix <directory> = Specify install directory
                                 [CMAKE_INSTALL_PREFIX].
  -Wdev                        = Enable developer warnings.
  -Wno-dev                     = Suppress developer warnings.
  -Werror=dev                  = Make developer warnings errors.
  -Wno-error=dev               = Make developer warnings not errors.
  -Wdeprecated                 = Enable deprecation warnings.
  -Wno-deprecated              = Suppress deprecation warnings.
  -Werror=deprecated           = Make deprecated macro and function warnings
                                 errors.
  -Wno-error=deprecated        = Make deprecated macro and function warnings
                                 not errors.
  --preset <preset>,--preset=<preset>
                               = Specify a configure preset.
  --list-presets               = List available presets.
[... 175 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/jcorc/hackrf/host/build

jcorc@Beast MINGW64 ~/hackrf/host/build
# make
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj
C:/msys64/home/jcorc/hackrf/host/libhackrf/src/hackrf.c:31:10: fatal error: libusb.h: No such file or directory
   31 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [libhackrf/src/CMakeFiles/hackrf.dir/build.make:76: libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: libhackrf/src/CMakeFiles/hackrf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

jcorc@Beast MINGW64 ~/hackrf/host/build
# make install
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj
C:/msys64/home/jcorc/hackrf/host/libhackrf/src/hackrf.c:31:10: fatal error: libusb.h: No such file or directory
   31 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [libhackrf/src/CMakeFiles/hackrf.dir/build.make:76: libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: libhackrf/src/CMakeFiles/hackrf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

corcorj, 2022-07-02:
that helped me overcome that specific line of code however when I moved onto the make and make install commands I got errors.. 

It looks like I didn't have libusb.h in the right directory on the make and make install commands but when I generated the make file with the command I was missing the $ with, looking at the output it does show that I have libusb installed correctly for that command to work correctly and output a make file or build files to : C:/msys64/home/jcorc/hackrf/host/build.

 I installed libusb with Cygwin and I also followed the link in the documentation listing it as a prereq to be able to install hackRF Software, however the download link downloaded a linux file and i am on windows the name of that file was: libusbx-1.0.18 (1).tar.bz2

Here is my commandline output:

jcorc@Beast MINGW64 ~
# cd hackrf

jcorc@Beast MINGW64 ~/hackrf
# cd host/build

jcorc@Beast MINGW64 ~/hackrf/host/build
# /usr/bin/cmake -version
bash: /usr/bin/cmake: No such file or directory

jcorc@Beast MINGW64 ~/hackrf/host/build
# /mingw64/bin/cmake -version
cmake version 3.23.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

jcorc@Beast MINGW64 ~/hackrf/host/build
# /mingw64/bin/cmake -h
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Options
  -S <path-to-source>          = Explicitly specify a source directory.
  -B <path-to-build>           = Explicitly specify a build directory.
  -C <initial-cache>           = Pre-load a script to populate the cache.
  -D <var>[:<type>]=<value>    = Create or update a cmake cache entry.
  -U <globbing_expr>           = Remove matching entries from CMake cache.
  -G <generator-name>          = Specify a build system generator.
  -T <toolset-name>            = Specify toolset name if supported by
                                 generator.
  -A <platform-name>           = Specify platform name if supported by
                                 generator.
  --toolchain <file>           = Specify toolchain file
                                 [CMAKE_TOOLCHAIN_FILE].
  --install-prefix <directory> = Specify install directory
                                 [CMAKE_INSTALL_PREFIX].
  -Wdev                        = Enable developer warnings.
  -Wno-dev                     = Suppress developer warnings.
  -Werror=dev                  = Make developer warnings errors.
  -Wno-error=dev               = Make developer warnings not errors.
  -Wdeprecated                 = Enable deprecation warnings.
  -Wno-deprecated              = Suppress deprecation warnings.
  -Werror=deprecated           = Make deprecated macro and function warnings
                                 errors.
  -Wno-error=deprecated        = Make deprecated macro and function warnings
                                 not errors.
  --preset <preset>,--preset=<preset>
                               = Specify a configure preset.
  --list-presets               = List available presets.
  -E                           = CMake command mode.
  -L[A][H]                     = List non-advanced cached variables.
  --build <dir>                = Build a CMake-generated project binary tree.
  --install <dir>              = Install a CMake-generated project binary
                                 tree.
  --open <dir>                 = Open generated project in the associated
[... 169 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/jcorc/hackrf/host/build

jcorc@Beast MINGW64 ~/hackrf/host/build
# make
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj
C:/msys64/home/jcorc/hackrf/host/libhackrf/src/hackrf.c:31:10: fatal error: libusb.h: No such file or directory
   31 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [libhackrf/src/CMakeFiles/hackrf.dir/build.make:76: libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: libhackrf/src/CMakeFiles/hackrf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

jcorc@Beast MINGW64 ~/hackrf/host/build
# make install
[  5%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj
C:/msys64/home/jcorc/hackrf/host/libhackrf/src/hackrf.c:31:10: fatal error: libusb.h: No such file or directory
   31 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [libhackrf/src/CMakeFiles/hackrf.dir/build.make:76: libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: libhackrf/src/CMakeFiles/hackrf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

stevewillson, 2022-07-02:
I'm not sure about using cygwin to install the libusb files.

I am using MSYS2 and used `pacman` to install the package `mingw-w64-x86_64-libusb` to provide the libusb files.

Here is a listing of the files that the package `mingw-w64-x86_64-libusb` provides:

`$ pacman -Ql PACKAGE` - list the files owned by the queried package (from `$ pacman -Q -h`)

```
$ pacman -Ql mingw-w64-x86_64-libusb
mingw-w64-x86_64-libusb /mingw64/
mingw-w64-x86_64-libusb /mingw64/bin/
mingw-w64-x86_64-libusb /mingw64/bin/libusb-1.0.dll
mingw-w64-x86_64-libusb /mingw64/include/
mingw-w64-x86_64-libusb /mingw64/include/libusb-1.0/
mingw-w64-x86_64-libusb /mingw64/include/libusb-1.0/libusb.h
mingw-w64-x86_64-libusb /mingw64/lib/
mingw-w64-x86_64-libusb /mingw64/lib/libusb-1.0.a
mingw-w64-x86_64-libusb /mingw64/lib/libusb-1.0.dll.a
mingw-w64-x86_64-libusb /mingw64/lib/pkgconfig/
mingw-w64-x86_64-libusb /mingw64/lib/pkgconfig/libusb-1.0.pc
mingw-w64-x86_64-libusb /mingw64/share/
mingw-w64-x86_64-libusb /mingw64/share/licenses/
mingw-w64-x86_64-libusb /mingw64/share/licenses/libusb/
mingw-w64-x86_64-libusb /mingw64/share/licenses/libusb/COPYING
```

The file `libusb.h` is located at: `/mingw64/include/libusb-1.0/libusb.h`

Do you have `libusb.h` at that same location?

To locate `libusb.h` use the `locate` command: `$ locate libusb.h`, but before doing that, update the local file index using `$ updatedb`

t812206236, 2023-01-29:
you use -DLIBUSB_INCLUDE_DIR to include lib path? that makes no sense,you should include header

joedwardson, 2024-03-20:
For anyone that encounter `[libusb ](ibusb.h: No such file or directory)` in the `make `step. Edit the import in source code helped for me:
**Open File->** C:\msys64\home\<checkyourpath>\hackrf\host\libhackrf\src\hackrf.c
**Change line** `#include <libusb.h>`
**To** `#include <libusb-1.0/libusb.h>`

With this fix I got passed this problem:
```
libusb.h: No such file or directory
31 | #include <libusb.h>
| ^~~~~~~~~~
```
I should mention I dont know if this is a valid fix, just wanted to post here in case someone else got stuck and want a quick fix.

shad0w82, 2024-06-12:
> For anyone that encounter `[libusb ](ibusb.h: No such file or directory)` in the `make `step. Edit the import in source code helped for me: **Open File->** [...]

The correct fix to overcome this problem, without modifying the source code, should be calling this as a CMAKE command:
`${MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=${MINGW_PREFIX}/include/libusb-1.0 -DFFTW_LIBRARIES=/lib
`

martinling, 2025-10-04:
Although it's possible to build with MSYS, at this point our normal method to build the HackRF software for Windows is using the MSVC compiler, with dependencies installed using `vcpkg`. This is the way we do it in our CI builds.

I've just opened PR #1595, which replaces all the old Windows build instructions in the documentation.

If people still find it useful to build with MSYS as an alternative, we could add a section on how to do that as well.

closed 2026-01-04 by mossmann