greatscottgadgets/hackrf #1110: Hissing/noise when creating fm radio receiver and have to reset hackrf after every execution of the flow graph

delliy, 2022-06-24:
### Steps to reproduce
1.I followed multiple tutorials to create an fm radio receiver including the one by Great Scott / Michael Ossmann
2.Also I downloaded some premade gnuradio fm radio project files that I found on here.
3.

### Expected behaviour
Should be able to hear local radio stations. 

### Actual behaviour
Despite following different tutorials and downloading project files each one came back with the same result. Hissy and indistinguishable noise being reproduced. I've tried tuning to local radio stations and still get the same result. On top of this if I try and execute the flow graph for a second time it receives absolutely no signal and the frequency display shows a flat line meaning I have to keep restarting the hackrf every time I want to execute the flow graph again. I've also tried extending the antenna and moving it in different positions and this didn't seem to help.

I'm using the ANT500

### Version information
Windows 10

**hackrf_info output:**
At certain frequency's it writes a bunch of OOOOO. Not sure if this is helpful

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GNUradio 3.8

### Output
```
Insert any commandline or build output here
```

delliy, 2022-06-26:
Small update, I managed to update the Hackrf to the latest 2021 firmware and still getting the same results even when using other programs such as SDRSharp.

straithe, 2022-06-26:
Can you provide links to the tutorials you are following?

delliy, 2022-06-27:
https://youtu.be/DAkh_lo1TO8

https://youtu.be/ye8wFVPF4wI

These two. But as I said I've also downloaded sdr sharp and I'm getting the same results. So my guess would be that it's a hardware/antenna issue rather than a software issue if I'm getting the same problem throughout multiple programs.

delliy, 2022-06-27:
So today I tried updating the libhackrf to see if that helped and now I'm getting a new error in gnuradio and now it won't even execute the flow graph.

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio v3.8.2.0-57-gd71cd177
built-in source types: file rtl rtl_tcp uhd hackrf bladerf airspy airspyhf soapy redpitaya 
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
Traceback (most recent call last):
  File "C:\Users\C\Documents\GNURadio\Test Projects\top_block.py", line 242, in <module>
    main()
  File "C:\Users\C\Documents\GNURadio\Test Projects\top_block.py", line 218, in main
    tb = top_block_cls()
  File "C:\Users\C\Documents\GNURadio\Test Projects\top_block.py", line 113, in __init__
    self.osmosdr_source_0 = osmosdr.source(
  File "C:\Program Files\GNURadio-3.8\lib\site-packages\osmosdr\osmosdr_swig.py", line 1316, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: Failed to open HackRF device (-1000) Operation not supported or unimplemented on this platform

>>> Done (return code 1)

A bit more information from running hackrf_info on my virtual linux machine

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: *
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: *

It seems weird to me that hackrf info version and libhackrf version is unknown. 

When I run the command ~$ sudo apt-get -y install libhackrf-dev it confirms that libhackrf is the latest version "libhackrf-dev is already the newest version (2021.03.1-2)."

straithe, 2022-06-27:
You now likely have a libhackrf and firmware mismatch since you've updated the firmware yourself and did libhackrf updates through apt-get. You'll want to make sure you are using the same source for both. 

The hackrf info version and libhackrf version being unknown is a known issue when updating through apt-get. 

Have you tried using your HackRF with GQRX? I suggest trying that (once you match your HackRF firmware and software) as a first step to seeing if it is the HackRF or software.

delliy, 2022-06-28:
I located the libhackrf folder in the latest release download and noticed that it contained the file 53-hackrf.rules. (I think I actually already did this manually to be able to install the firmware but then potentially overwrote it with the apt-get function?) Anyway, I overwrote the rules using the command "sudo nano /etc/udev/rules.d/53-hackrf.rules" and pasted in the code from the 53-hackrf.rules file that came with the download and I was still getting the unknown message for the hackrf_info and libhackrf version. I noticed two other files in the folder named CmakeLists.txt and libhackrf.pc.in so tried to search online what to do with these and read about installing from the source using this code: 

cd hackrf/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

I'm not exactly sure how it knows where to extract this information from but now when I run hackrf_info I get

hackrf_info version: git-34cbbe7
libhackrf version: git-34cbbe7
Found HackRF
Index: 0
Serial number: *
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: *

This still seems like a mismatch in firmware and the libhackrf version? I also couldn't really find much information online about the libhackrf.pc.in file either. Really sorry but I've never used linux before so trying my best to figure this out.

straithe, 2022-07-14:
If you are just learning about your HackRF I suggest trying GQRX first to listen to signals. After that I suggest trying the GNU Radio tutorials (https://wiki.gnuradio.org/index.php/Tutorials) and then trying the tutorials from Mossmann and Great Scott Gadgets (youtube.com/greatscottgadgets).

straithe, 2022-08-13:
I'm going to close this as there hasn't been a response in the last thirty days, but please re-open this issue or open a new one if you still need assistance.

closed 2022-08-13 by straithe