greatscottgadgets/hackrf #1111: Hack RF one with Rational waves issue

jhorton845, 2022-06-25:
### Steps to reproduce
1. Downloaded Rational Waves software
2. Did driver install when prompted - everything says successful
3. Launch software

### Expected behaviour
software should open and recognize hack rf one.

### Actual behaviour
software scans for hack rf one and only launches in demo mode

### Version information
**Operating system**:
Windows 11

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
Rational Waves software by nutsaboutnets

### Output
```
Insert any commandline or build output here
```

straithe, 2022-06-26:
Can you provide a link to the Rotational Waves software you downloaded?

jhorton845, 2022-06-26:
https://nutsaboutnets.com/rationalwaves/

My link came directly from a purchase. When they send you a license.

straithe, 2022-06-26:
I do not have experience this software. Have you contacted nutsaboutnets regarding this issue?

jhorton845, 2022-06-26:
Yes.. still waiting on a reply from them. Been about a week now. 

Justin Horton
Monitor Engineer
Audio Systems Tech 
- New York -
845.505.7154

straithe, 2022-06-26:
Since Rational Waves is closed source software nutsaboutnets is likely your best option for help. The best I can suggest is that you look for somewhere to input a license key in the demo mode software. Good luck!

closed 2022-07-21 by straithe